exelban/stats #203: restart app when it update, the menu bar will be duplicate.

crownzdp, 2020-12-01:


geraldlnj, 2020-12-01:
I had this issue too. In my case, I had one instance of Stats running from Application folder, and one instance running from the disk image.

crownzdp, 2020-12-01:
Just one instance,maybe it has an old instance and a new instance when update finished?

exelban, 2020-12-01:
Hi. It could happen from time to time. It was the first update of the Stats? Or you normally update the Stats previously and only the last update runs in 2 instances (do not close the first one)?

crownzdp, 2020-12-02:
no, it happened before，but I just tested it , it's ok.
last update finished, menu bar didn't show automatically, and it start it manual, it start two instance, but this time it shows automatically.

closed 2020-12-04 by crownzdp