exelban/stats #204: Sensors doesn't show up in dark mode

bigbang11, 2020-12-01:
Hi, in the actual version of stats, sensors are invisible in the menu task. I use big sur in dark mode. other tasks are ok.

[image]
[image]

exelban, 2020-12-01:
Hi. Thanks for reporting.
Do you use multi-display setup?

bigbang11, 2020-12-01:
Hi you are Welcome no I don’t Use multipel Displays

Mit freundlichen Grüßen
Baris Doganay

exelban, 2020-12-01:
And in the light mode, the values are visible?

ipatterson, 2020-12-02:
[image]

Not OP, but I wanted to comment that I'm seeing the same issue in Light Mode

exelban, 2020-12-02:
@ipatterson Thanks. It looks like the same issue. And if you enable only 1 sensor? Will it shows?
And what if you enable more than 2?

ipatterson, 2020-12-02:
So it does resolve the display issue if sensors are manually toggled. Enabling one sensor shows up in the menu bar. Two, on multiple lines. Four, it extends:

[image]

And for reference, this is the default setting I had when I responded to this thread:

[image]

exelban, 2020-12-02:
Em. Wat?
You have disabled all sensors, and expecting that something will be visualized?

exelban, 2020-12-02:
@bigbang11 Could you please check if you have enabled at least 1 sensor?

ipatterson, 2020-12-02:
When I have one sensor enabled, I do see it on the menu bar

exelban, 2020-12-02:
@ipatterson and do not see if more than 1 enabled?

exelban, 2020-12-06:
Ok. I suppose @bigbang11 and @ipatterson do not have enabled at least 1 sensor. That was the reason that sensors were not visible in the menu bar. If not, reopen the issue please.

closed 2020-12-06 by exelban