exelban/stats #210: Sensors Not Displaying

ChasingDopamine, 2020-12-03:
Sensors not properly displaying.  I originally got them to work on first install, but upon starting the system the following day the temperature sensors were no longer displaying.

- When "Sensors" toggled ON, an interactive blank section appears in the menu bar where the display should be.
- Upon clicking blank area, the entire sensors dropdown is blank as well.  See image below. _(the temp display next to it is TG Pro).  That's just there temporarily as a test to see if other apps displayed the proper information._
[image]

What I've tried:

- Enabling permissions for Full Disk Access, Accessability.
- Toggling permissions on and off.
- Restarting the application and MacBook.
- Fully remove/uninstall Stats and reinstall

Information:
macOS Big Sur 11.0.1
MacBook Pro (M1, 2020)
Stats version 2.4.6

johnalanwoods, 2020-12-03:
Same issue built from src with current head.

lucasaf04, 2020-12-03:
Duplicate of #204

exelban, 2020-12-03:
Hi. Thanks for opening the issue. This problem was described [here](https://github.com/exelban/stats/issues/181).
TL;DR: Computers with Apple Silicon has different sensors key that Intel based. And for now, there is no mapping of these keys.

@lucasaf04 nope. There are no relations between these issues.

thebitstick, 2020-12-05:
I have access to the Sensei app which has a very rich list of sensors.
[image]
[image]
[image]

_Originally posted by @thebitstick in https://github.com/exelban/stats/issues/181#issuecomment-738955290_

exelban, 2020-12-05:
@thebitstick Hi. Thanks for the screenshots.
This app just shows the keys, not sensor names. So it's not really helpful(

johnalanwoods, 2020-12-10:
Can you clarify what "keys" and "names" are?

exelban, 2020-12-10:
Yeap.

In the SMC there is only keys and values. So I can receive all possible keys + values which present on the system. This list will contains sensors, some system specs and some data (timestamp of last boot) etc. The key it's a 4 chars. When the list is received I must to filter it. To have the list with interesting key value pair. It's a pretty easy. Because normally the first char show the type of value. `TC0D` - means temperature on core.
After filtering I don't want to see the list of key values. I want to see the name of the sensor and it's value.
To do this, I need the dictionary. Which can translate the key to sensor name.

TL.DR: You could see an example of the key and name [here](https://github.com/exelban/stats/blob/6b36d3e9773d3fa2c3fd3c6047227a55d16ea5e8/Modules/Sensors/values.swift#L101-L176).

johnalanwoods, 2020-12-10:
@exelban OK that makes sense, so you need help mapping the key (`TC0D`), to the actual sensor name?

I wonder where other apps got their mapping, I guess I could try and reverse engineer in Ghidra.

exelban, 2020-12-10:
Yes. 

I don't see any app that has a dictionary for Apple Silicon (or maybe there is some, I don't have Apple Silicon).
Unfortunately, it's not public information. If even some company has the dictionary for m1. They will not share. Because they it give them an advantage.
I suppose the only one which can have the good dictionary it's iStat Menus.

Fun fact. There is no complete list even for intel based mac(

andreialecu, 2020-12-18:
Perhaps this helps? https://github.com/fermion-star/apple_sensors

exelban, 2020-12-18:
Thanks. It could be helpful.
If I understand correctly, someone must run this tool on m1 and post the output.

lucasaf04, 2020-12-18:
A bit weird output:

`./temp_sensor`

```
PMU tdie7, PMU2 TR4b, PMU tdev4, PMU tdev7, PMU tdie5, PMU tdev1, gas gauge battery, NAND CH0 temp, PMU2 TR3d, PMU2 TR6b, gas gauge battery, PMU2 TR8b, PMU tdie8, PMU TP3w, PMU tdev5, PMU2 TR1l, gas gauge battery, PMU tdie1, PMU2 TR0Z, pACC MTR Temp Sensor4, PMU tdev8, PMU2 TR3b, PMU tdev2, PMU tdie4, SOC MTR Temp Sensor0, PMU tdie2, PMU tcal, gas gauge battery, PMU2 TR2d, pACC MTR Temp Sensor2, PMU2 TR5d, PMU2 TR5b, ANE MTR Temp Sensor1, eACC MTR Temp Sensor0, PMGR SOC Die Temp Sensor2, PMU2 TR4d, PMU tdev6, eACC MTR Temp Sensor3, GPU MTR Temp Sensor1, PMU2 TR7b, pACC MTR Temp Sensor9, PMU tdie6, SOC MTR Temp Sensor2, ISP MTR Temp Sensor5, PMGR SOC Die Temp Sensor1, PMU tdev3, PMGR SOC Die Temp Sensor0, gas gauge battery, pACC MTR Temp Sensor7, pACC MTR Temp Sensor8, SOC MTR Temp Sensor1, pACC MTR Temp Sensor3, PMU2 TR2l, pACC MTR Temp Sensor5, gas gauge battery, PMU2 TR1d, GPU MTR Temp Sensor4, 
33.1, 34.8, -21.4, 29.8, 33.4, 30.4, 27.9, 28.0, 29.7, 35.6, 27.9, 36.0, 32.7, 35.8, -21.5, 36.0, 27.9, 36.4, 51.9, 28.9, 31.1, 33.9, 29.6, 34.8, 27.3, 32.1, 51.9, 27.0, 29.8, 27.4, 28.8, 36.0, 30.0, 27.0, 27.9, 29.0, 30.1, 24.8, 30.0, 34.8, 29.0, 34.7, 30.7, 30.0, 28.0, 29.8, 28.0, 27.0, 28.9, 26.9, 26.5, 28.8, 35.9, 27.3, 27.7, 29.9, 30.0,
```

`./temp_sensor | ./monitor`

```
ANE MTR Temp Sensor1       |  
GPU MTR Temp Sensor1       |  
GPU MTR Temp Sensor4       |  
ISP MTR Temp Sensor5       |  
NAND CH0 temp              |  
PMGR SOC Die Temp Sensor0  |  
PMGR SOC Die Temp Sensor1  |  
PMGR SOC Die Temp Sensor2  |  
PMU TP3w                   |  
PMU tcal                   |  
PMU tdev1                  |  
PMU tdev2                  |  
PMU tdev3                  |  
PMU tdev4                  |  
PMU tdev5                  |  
PMU tdev6                  |  
PMU tdev7                  |  
PMU tdev8                  |  
PMU tdie1                  |  
PMU tdie2                  |  
PMU tdie4                  |  
PMU tdie5                  |  
PMU tdie6                  |  
PMU tdie7                  |  
PMU tdie8                  |  
PMU2 TR0Z                  |  
PMU2 TR1d                  |  
PMU2 TR1l                  |  
PMU2 TR2d                  |  
PMU2 TR2l                  |  
PMU2 TR3b                  |  
PMU2 TR3d                  |  
PMU2 TR4b                  |  
PMU2 TR4d                  | 30.0
PMU2 TR5b                  | 30.0
PMU2 TR5d                  | 30.0
PMU2 TR6b                  | 30.0
PMU2 TR7b                  | 28.0
PMU2 TR8b                  | 29.1
SOC MTR Temp Sensor0       | 29.0
SOC MTR Temp Sensor1       | 29.0
SOC MTR Temp Sensor2       | 36.2
eACC MTR Temp Sensor0      | 51.9
eACC MTR Temp Sensor3      | 30.6
gas gauge battery          | 30.2
gas gauge battery          | 30.1
gas gauge battery          | -21.4
gas gauge battery          | -21.6
gas gauge battery          | 30.3
gas gauge battery          | 30.4
pACC MTR Temp Sensor2      | 31.5
pACC MTR Temp Sensor3      | 36.0
pACC MTR Temp Sensor4      | 33.4
pACC MTR Temp Sensor5      | 35.0
pACC MTR Temp Sensor7      | 35.1
pACC MTR Temp Sensor8      | 34.8
pACC MTR Temp Sensor9      | 35.0
                             33.4
                             51.9
                             30.5
                             36.6
                             30.1
                             36.4
                             33.9
                             30.1
                             36.4
                             29.2
                             36.4
                             29.0
                             36.4
                             36.5
                             36.2
                             28.5
                             28.4
                             31.5
                             26.9
                             25.5
                             28.0
                             28.0
                             28.0
                             27.0
                             28.0
                             27.8
                             28.2
                             32.4
                             31.1
                             27.5
                             30.0
                             30.4
                             31.2
```

exelban, 2020-12-18:
Yeap, the output is a little bit strange. Also the keys are missing( But I see some sensors which could be added to the Stats.
I will try to make a little bit changes to this `temp_sensor` app to show key - name.

andreialecu, 2020-12-20:
Here's the output on my Mac Mini M1:

```
apple_sensors on  master [?] took 37s 
➜ ./temp_sensor | ./monitor
ANE MTR Temp Sensor1       | 30.0
GPU MTR Temp Sensor1       | 27.0
GPU MTR Temp Sensor4       | 27.6
ISP MTR Temp Sensor5       | 30.0
NAND CH0 temp              | 28.0
PMGR SOC Die Temp Sensor0  | 29.0
PMGR SOC Die Temp Sensor1  | 28.7
PMGR SOC Die Temp Sensor2  | 28.5
PMU TP3w                   | 37.1
PMU tcal                   | 51.9
PMU tdev1                  | 29.9
PMU tdev2                  | 30.3
PMU tdev3                  | 30.5
PMU tdev4                  | -21.5
PMU tdev5                  | -21.5
PMU tdev6                  | 29.5
PMU tdev7                  | 33.4
PMU tdev8                  | 31.3
PMU tdie1                  | 37.3
PMU tdie2                  | 34.3
PMU tdie4                  | 36.5
PMU tdie5                  | 36.2
PMU tdie6                  | 36.0
PMU tdie7                  | 35.9
PMU tdie8                  | 34.4
PMU2 TR0Z                  | 51.9
PMU2 TR1d                  | 31.2
PMU2 TR1l                  | 37.1
PMU2 TR2d                  | 29.2
PMU2 TR2l                  | 38.0
PMU2 TR3b                  | 36.8
PMU2 TR3d                  | 29.4
PMU2 TR4b                  | 37.4
PMU2 TR4d                  | 29.6
PMU2 TR5b                  | 37.8
PMU2 TR5d                  | 31.2
PMU2 TR6b                  | 37.0
PMU2 TR7b                  | 38.2
PMU2 TR8b                  | 36.8
SOC MTR Temp Sensor0       | 26.9
SOC MTR Temp Sensor1       | 28.0
SOC MTR Temp Sensor2       | 28.6
eACC MTR Temp Sensor0      | 27.0
eACC MTR Temp Sensor3      | 26.7
pACC MTR Temp Sensor2      | 30.8
pACC MTR Temp Sensor3      | 26.4
pACC MTR Temp Sensor4      | 27.1
pACC MTR Temp Sensor5      | 29.0
pACC MTR Temp Sensor7      | 27.7
pACC MTR Temp Sensor8      | 27.5
pACC MTR Temp Sensor9      | 26.6
```

@lucasaf04 I noticed that if you run it with a terminal window size that is not High enough, it can glitch, like your output. Make sure to resize the terminal and run it again and everything should line up.

lucasaf04, 2020-12-20:
@andreialecu Thanks, I have updated my comment.

andreialecu, 2020-12-20:
I've been running https://www.tunabellysoftware.com/cpu_stress_test/ to see how the numbers change, and after a few minutes here are the results:

```
➜ ./temp_sensor | ./monitor
ANE MTR Temp Sensor1       | 30.0
GPU MTR Temp Sensor1       | 42.1
GPU MTR Temp Sensor4       | 45.9
ISP MTR Temp Sensor5       | 30.0
NAND CH0 temp              | 35.0
PMGR SOC Die Temp Sensor0  | 43.0
PMGR SOC Die Temp Sensor1  | 46.4
PMGR SOC Die Temp Sensor2  | 54.5
PMU TP3w                   | 57.9
PMU tcal                   | 51.9
PMU tdev1                  | 42.6
PMU tdev2                  | 51.8
PMU tdev3                  | 32.8
PMU tdev4                  | -21.7
PMU tdev5                  | -21.6
PMU tdev6                  | 31.7
PMU tdev7                  | 39.6
PMU tdev8                  | 48.6
PMU tdie1                  | 58.2
PMU tdie2                  | 66.6
PMU tdie4                  | 59.6
PMU tdie5                  | 63.5
PMU tdie6                  | 59.4
PMU tdie7                  | 65.1
PMU tdie8                  | 59.7
PMU2 TR0Z                  | 51.9
PMU2 TR1d                  | 34.7
PMU2 TR1l                  | 45.6
PMU2 TR2d                  | 30.7
PMU2 TR2l                  | 46.0
PMU2 TR3b                  | 44.7
PMU2 TR3d                  | 34.9
PMU2 TR4b                  | 45.6
PMU2 TR4d                  | 35.1
PMU2 TR5b                  | 44.8
PMU2 TR5d                  | 38.5
PMU2 TR6b                  | 45.6
PMU2 TR7b                  | 45.7
PMU2 TR8b                  | 44.7
SOC MTR Temp Sensor0       | 42.3
SOC MTR Temp Sensor1       | 47.3
SOC MTR Temp Sensor2       | 55.2
eACC MTR Temp Sensor0      | 49.5
eACC MTR Temp Sensor3      | 46.2
pACC MTR Temp Sensor2      | 67.5
pACC MTR Temp Sensor3      | 71.5
pACC MTR Temp Sensor4      | 67.3
pACC MTR Temp Sensor5      | 56.2
pACC MTR Temp Sensor7      | 64.8
pACC MTR Temp Sensor8      | 64.1
pACC MTR Temp Sensor9      | 61.1
```

Seems like most of the impact is on the `PMU tdie*` (perhaps temp for each core?) and `pACC MTR Temp Sensor*` values.

exelban, 2020-12-20:
There is one big problem for me with this output. There are no SMC keys( 
Only a few of them `TRxb`.

andreialecu, 2020-12-20:
@exelban I'm not sure if I'm understanding exactly how this works. But from what I can tell Apple Silicon does not use (?) SMC for these, but you need to use the HID apis instead.

Based on the output above I think that `eACC MTR Temp Sensor*` is the temperature of the efficiency cores, and `pACC MTR Temp Sensor*` is for the performance cores. It seems to correlate with what other apps that display temperatures show.

exelban, 2020-12-20:
What do you mean `Apple Silicon does not use SMC`?

andreialecu, 2020-12-20:
Not sure if this is relevant, but the repo above mentions running `ioreg -lfx`. I'm attaching the output that seems related to sensors here:

https://pastebin.com/raw/BywBmuFP

Perhaps this helps.

Regarding SMC, I'm not sure what I mean. I remember reading somewhere that the sensors are not available via SMC, or it makes it too complicated.

Why would the SMC keys be necessary when the code here seems to work without them? https://github.com/fermion-star/apple_sensors/blob/master/temp_sensor.m

exelban, 2020-12-20:
How access to the sensors via `ioreg` solve the problem?

andreialecu, 2020-12-20:
I don't claim to know anything about this really, it's just something in the readme:

https://github.com/fermion-star/apple_sensors
> Only test with my Mac mini with M1. Please check your mac's ioreg -lfx output to make changes in temp_sensor.m if needed.

exelban, 2020-12-20:
Read the next line please:
`For better names (e.g. what is PMU TP3w ?) for the sensors, please refer to...`

andreialecu, 2020-12-20:
I see, but I don't think anyone really figured out what they mean. Everyone is just guessing. See: https://forums.macrumors.com/threads/istat-menus-now-sort-of-supports-temperature-sensors-on-m1.2276062/

exelban, 2020-12-20:
I read this. I still do not understand what do you propose?

andreialecu, 2020-12-20:
I guess I don't even understand what the problem is. Care to elaborate? :)

andreialecu, 2020-12-20:
If the problem is not being able to associate the "raw" names with more user-friendly names, then I don't think there's much that can be done about that at the moment. 

Other temperature apps seem to either display the raw names, or they're guessing that eACC = efficiency packages, and pACC = performance package.

Might still be useful to display them in their raw form, and update it later to more friendly names, after it can be figured out exactly what they mean.

exelban, 2020-12-20:
I prefer to wait till some dictionary will be ready. Then show raw names.
For you maybe it's ok to see `PMU TP3w = 24C`. Because you realizing the problem. But for most people, it will look like a bug.

exelban, 2021-02-01:
Hi.
I have an ask. Could maybe someone who have m1 macbook share one internal library. In theory it must be the same for all macs. But maybe m1 have a different one. This lib has the naming of sensors. Which will be really helpfull to this issue)

I'm talking about `libSMC.dylib`. Which you can find here: `/usr/lib/libSMC.dylib`.

Wilkolicious, 2021-02-01:
@exelban Don't have that unfortunately
```
❯ l /usr/lib
total 4672
drwxr-xr-x  37 root  wheel   1.2K  1 Jan  2020 .
drwxr-xr-x@ 11 root  wheel   352B  1 Jan  2020 ..
-r--r--r--   1 root  wheel   198B  1 Jan  2020 charset.alias
lrwxr-xr-x   1 root  wheel    12B  1 Jan  2020 cron -> ../../var/at
-rwxr-xr-x   1 root  wheel   539K  1 Jan  2020 dsc_extractor.bundle
drwxr-xr-x  15 root  wheel   480B  1 Jan  2020 dtrace
-rwxr-xr-x   1 root  wheel   2.4M  1 Jan  2020 dyld
drwxr-xr-x   4 root  wheel   128B  1 Jan  2020 groff
-rwxr-xr-x   1 root  wheel   135K  1 Jan  2020 libLeaksAtExit.dylib
-rwxr-xr-x   1 root  wheel   2.7M  1 Jan  2020 libMTLCapture.dylib
-rwxr-xr-x   1 root  wheel   102K  1 Jan  2020 libffi-trampolines.dylib
-rwxr-xr-x   1 root  wheel   173K  1 Jan  2020 libgmalloc.dylib
lrwxr-xr-x   1 root  wheel    27B  1 Jan  2020 libhunspell-1.2.0.dylib -> libhunspell-1.2.0.0.0.dylib
lrwxr-xr-x   1 root  wheel    27B  1 Jan  2020 libhunspell-1.2.dylib -> libhunspell-1.2.0.0.0.dylib
lrwxr-xr-x   1 root  wheel    21B  1 Jan  2020 libiodbc.2.dylib -> libiodbc.2.1.18.dylib
lrwxr-xr-x   1 root  wheel    21B  1 Jan  2020 libiodbc.dylib -> libiodbc.2.1.18.dylib
lrwxr-xr-x   1 root  wheel    25B  1 Jan  2020 libiodbcinst.2.dylib -> libiodbcinst.2.1.18.dylib
lrwxr-xr-x   1 root  wheel    25B  1 Jan  2020 libiodbcinst.dylib -> libiodbcinst.2.1.18.dylib
-rwxr-xr-x   1 root  wheel   262K  1 Jan  2020 libobjc-trampolines.dylib
lrwxr-xr-x   1 root  wheel    18B  1 Jan  2020 libpython.dylib -> libpython2.7.dylib
lrwxr-xr-x   1 root  wheel    68B  1 Jan  2020 libpython2.7.dylib -> ../../System/Library/Frameworks/Python.framework/Versions/2.7/Python
lrwxr-xr-x   1 root  wheel    21B  1 Jan  2020 libstdc++.6.dylib -> libstdc++.6.0.9.dylib
lrwxr-xr-x   1 root  wheel    17B  1 Jan  2020 libstdc++.dylib -> libstdc++.6.dylib
drwxr-xr-x   2 root  wheel    64B  1 Jan  2020 log
drwxr-xr-x  20 root  wheel   640B  1 Jan  2020 pam
drwxr-xr-x   4 root  wheel   128B  1 Jan  2020 php
drwxr-xr-x   7 root  wheel   224B  1 Jan  2020 pkgconfig
lrwxr-xr-x   1 root  wheel    75B  1 Jan  2020 python2.7 -> ../../System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7
drwxr-xr-x   9 root  wheel   288B  1 Jan  2020 rpcsvc
lrwxr-xr-x   1 root  wheel    76B  1 Jan  2020 ruby -> ../../System/Library/Frameworks/Ruby.framework/Versions/Current/usr/lib/ruby
drwxr-xr-x  22 root  wheel   704B  1 Jan  2020 sasl2
lrwxr-xr-x   1 root  wheel    27B  1 Jan  2020 sqlite3 -> /System/Library/Tcl/sqlite3
drwxr-xr-x   7 root  wheel   224B  1 Jan  2020 swift
drwxr-xr-x   7 root  wheel   224B  1 Jan  2020 system
drwxr-xr-x   2 root  wheel    64B  1 Jan  2020 updaters
drwxr-xr-x   3 root  wheel    96B  1 Jan  2020 xpc
drwxr-xr-x   3 root  wheel    96B  1 Jan  2020 zsh
```

RajceP, 2021-02-05:
Libraries are not present in file system in Big Sur. They have moved to shared cache.

Jethro1000, 2021-02-12:
Could it be possible to have as a minimum the actual Fan speed?
FNum (number of Fans) and F0Ac (Fan 0 Actual speed) is working

nik27, 2021-02-23:
> Hi.
> I have an ask. Could maybe someone who have m1 macbook share one internal library. In theory it must be the same for all macs. But maybe m1 have a different one. This lib has the naming of [...]

I just checked and there is libSMC.dylib but as @RajceP pointed out location was changed in BigSur. After extracting shared cache with dyld_shared_cache_util I found it in `/usr/lib/libSMC.dylib`.

FireLite4, 2021-03-04:
[image]

-sudo spindump
-cat /tmp/spindump.txt | grep "Fan speed"

I am Running an m1 MBP (I used tg pro to set the fan speed)

exelban, 2021-03-04:
@FireLite4 Thanks for the proposal. But: 1 - it's an external tool, 2 - requires `sudo`.

IamTheLoki, 2021-03-21:
Is this Issue closed forever or is there a chance this feature can be implement in a future version?

exelban, 2021-03-21:
Hi. Yes, it will be implemented. I need an M1 device to do it. I have only intel-based. And there is a difference in fetching sensors. Or someone with m1 and knowledge in swift could help.

cristiansori2, 2021-03-22:
> Hi. Yes, it will be implemented. I need an M1 device to do it. I have only intel-based. And there is a difference in fetching sensors. Or someone with m1 and knowledge in swift could help.

I have a MP with M1,and I can help.Best regards.

exelban, 2021-03-24:
@cristiansori2 thanks. Do you have experience with swift?

cristiansori2, 2021-03-24:
> @cristiansori2 thanks. Do you have experience with swift?

Yes,what do you need?

exelban, 2021-03-24:
Ok. I have already tried to port [this](https://github.com/fermion-star/apple_sensors/blob/master/temp_sensor.m) and [this](https://github.com/yujitach/MenuMeters/blob/master/hardware_reader/applesilicon_hardware_reader.m). 
But I receive an empty list on intel. So, I'm stuck a little bit. 

I will put all my tests to the swift playground and create a separate repo for it. I will try to make it later today. And post here a link. So everyone could help.

cristiansori2, 2021-03-24:
> Ok. I have already tried to port [this](https://github.com/fermion-star/apple_sensors/blob/master/temp_sensor.m) and [...]

Ok pls let me know when you do it.

exelban, 2021-03-24:
I will try to rent an M1 (mac mini) as a Service.
I don't think that creating a separate repo for this it's a good idea. And don't see a better way to work with this. But it will only if I can't rent M1.

FireLite4, 2021-03-25:
it seems to be working
[image]

however the sensors tab and the fans tab has been deactivated in the apple silicon version so the only way to get the readout is to run it with rosetta.

there is also a small problem when the fan RPM goes to Zero the widget does not update and stays on the last update
[image]

exelban, 2021-03-25:
@FireLite4 yes. It works with rosetta. Because rosetta simulates some intel-based methods (SMC). But it does not work natively.

IamTheLoki, 2021-03-25:
Maybe you can mark this feature as experimental until a native solution is ready?

exelban, 2021-03-25:
Can someone run this console app?
It will create a file `log.txt` in the Documents. There must be thermal sensors. And share the log file after.

[m1_sensors.zip](https://github.com/exelban/stats/files/6208114/m1_sensors.zip)

IamTheLoki, 2021-03-25:
Yes

[log.txt](https://github.com/exelban/stats/files/6208145/log.txt)

exelban, 2021-03-25:
@IamTheLoki Thanks for this. It looks like I have a list of sensors. Now I need to figured out with values.

exelban, 2021-03-25:
A small fix. It must have values or give info on why there are no values.

[m1_sensors.zip](https://github.com/exelban/stats/files/6208197/m1_sensors.zip)

IamTheLoki, 2021-03-25:
[log.txt](https://github.com/exelban/stats/files/6208240/log.txt)

exelban, 2021-03-25:
Nice. It looks like we have sensor name and value)
Thanks for help.

exelban, 2021-03-25:
It will be not released this week. Because for this release I have already enough changes and new features. So it will be available only after 1 release.

But I will post here a build with this feature when it will be available.

andreialecu, 2021-03-26:
@exelban looks to me that the names of the sensors are exactly the same as the ones here: https://github.com/exelban/stats/issues/210#issuecomment-748612663

You mentioned back then that these names are unusable. I was wondering what changed.

exelban, 2021-03-26:
There is only one way to fetch sensors on M1. I was thinking that m1 still uses SMC. But not.
So there is no other way than to have these strange names.

I hope that with time we will fill mapping of sensor names)

exelban, 2021-03-30:
Hi.
There is the first build with a new sensor reader for apple silicon. Please give me know if it works. And if there are bugs.

[Stats.dmg.zip](https://github.com/exelban/stats/files/6232180/Stats.dmg.zip)

RajceP, 2021-03-30:
> Hi.
> There is the first build with a new sensor reader for apple silicon. Please give me know if it works. And if there are bugs.
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6232180/Stats.dmg.zip)

Unfortunately the widget doesn't show in menu bar.

exelban, 2021-03-30:
There is anything in the settings? Or you don't see even Sensors settings?

RajceP, 2021-03-30:
> There is anything in the settings? Or you don't see even Sensors settings?

Everything looks as expected in settings. But I cant see the widget after enabling the master sensors switch and some individual sensors. I can see the text (SEN) widget if enabled.

exelban, 2021-03-30:
So you have a list of sensors in the settings?

[36 more replies]

closed 2021-04-11 by exelban