exelban/stats #213: Feature request: Charging indicator for Battery

iliakukharev, 2020-12-06:
Would be great to see the lightning icon on the left of Battery "56%", like "⚡56%" when MacBook is charging.

exelban, 2020-12-06:
Hi. Thanks for the feedback.

There is a lightning icon when the battery widget selected.

iliakukharev, 2020-12-06:
Yes, but not when the "Numbers" mode selected
[image]

exelban, 2020-12-06:
Oh, I see.
Unfortunately, it's not possible to add lighting to this widget. The reason is simple. All text widgets must look the same. Also, it cannot be lighting as emoji. Because it will look too unnatural in the menu bar.

The only way to add some lightning to this view. It's writing a new widget for the battery and draw manually lightning. But I don't want to provide too many different widgets. It allows reducing bugs and makes support easier.

Sorry, but it cannot be implemented.

iliakukharev, 2020-12-06:
What about any other notification like `*56%`? Or `56%*`

exelban, 2020-12-06:
Unfortunately in this widget, there is no option to add something.
If I add some symbol like `*` to show the charging state. After releasing someone will ask to "fix" with lightning or plug)
And I will have no option after. Because `*` does not have sense even for me.

exelban, 2020-12-06:
I would really recommend switching to the battery widget. It can show not only the percentage but also the time to charge/discharge.

iliakukharev, 2020-12-06:
Thank you for the explanation.

As for the widget, I like this one, but it does not have %
[image]

iliakukharev, 2020-12-06:
ah sorry. I'm blind. didn't see the options.
battery icon with additional info is the perfect option.
you can close this request :)

closed 2020-12-06 by exelban