exelban/stats #214: Incorrect sensor names for different versions of macs

Military-Doctor, 2020-12-07:
**Describe the bug**

The sensor of _**heat-pipe**_ and _**CPU Core**_ should start with "1" rather than "2". 

Maybe the same SMC Key behaves differently on different machines?

**Details:**
 - Device: MacBook Pro 16" 2019
 - MacOS 10.15.7 (19H15)
 - Application version 2.4.7

[image]

After review `SMCSensorKeys.txt` mentioned in the code, I found that the SMC key "`TC1C`" could be `CPU Core 0 temperature` or `CPU Core 1 temperature`.

```
[TC1C] type [sp7s] 73703738 len [ 2]: CPU Core 0 temp (PECI) (DegC) (TC1C)
[TC1C] type [sp7s] 73703738 len [ 2]: CPU Core 1 absolute cooked temp (PECI) (DegC) (TC1C)
[TC1C] type [sp7s] 73703738 len [ 2]: CPU Core 1 temp (PECI) (DegC) (TC1C)
[TC1c] type [sp7s] 73703738 len [ 2]: CPU Core 1 absolute raw temp (PECI) (DegC) (TC1c)
[TC1C] type [sp7s] 73703738 len [ 2]: Max of CPU Core vs. Mem Ctlr/IG PECI Temps(TC1C)
[TC1C] type [sp7s] 73703738 len [ 2]: Max of CPU Core vs. Mem Ctlr/IG PECI temps (DegC) (SIS! force bit 5) (TC1C)
```

Then, I found this document, it shows that same SMC keys **could have** different meanings on various machines.

https://github.com/acidanthera/VirtualSMC/blob/master/Docs/iStat.txt

**For example:** 

Mac Mini (Late 2012), `TC1C` means the temperature of `CPU Core 2 - DTS`
However, in MacBook Pro (15 inch, Retina & Touch Bar, Mid 2017), `TC1C` means the temperature of `CPU Core 1`

Military-Doctor, 2020-12-07:
In repo https://github.com/acidanthera/VirtualSMC

They fixed this problem in release [1.0.2](https://github.com/acidanthera/VirtualSMC/compare/1.0.1...1.0.2) 

They said:

> Fixed TC0C/TC1C selection in SMCProcessor to match Mac models

I think maybe we can fix this problem by introducing a document that describe the specific SMC keys.

But this is a very tedious work. 😅

Military-Doctor, 2020-12-07:
Another reference I found: https://github.com/acidanthera/VirtualSMC/blob/master/Tools/CoreOffset/main.swift

They called it `CoreOffset`

Maybe we can introduce both `CoreOffset` and `TemperatureOffset`

exelban, 2020-12-07:
Hi.

There is no problem with naming `Core Offset` or `TemperatureOffset`.
There is a different problem. On some old macs the CPU core starts counting from `TC0C `. And on the newest from `TC1C`.

I don't know what the good way to resolve it now(

Military-Doctor, 2020-12-07:
An inelegant solution is to renumber the sensors name, because we are not affected by these offsets.

exelban, 2020-12-07:
Yeap. You are right.
It looks like I change the numbers in [this](https://github.com/exelban/stats/commit/133eff398ff703182343035eec20f0b0b1b8c8f7#diff-5b0d81805b21438e7281098949aaca0f4bedac21fdfdace30eb6f87ae950bfaa) commit.

I was supposed that I missed the numbers before.

It looks like it must be reverted to the original counting `TC1C = CPU core 1`.

exelban, 2020-12-07:
Hmm, I only now check my sensors. And if I revert the changes (`TC1C = CPU core 1`). I will have the wrong naming. 
Because now I have all cores displayed correctly.

[image]

exelban, 2020-12-08:
But from another hand. On my MacBook. I have the same problem as you.

[image]

Military-Doctor, 2020-12-08:
Well, you don't need to revert to the original counting,

Here is my solution:

```python
if ( the first cpu sensor string ends with "2") : # offset occurs, all sensor must starts from 1
    renumber_the_sensors()

if ( the first temperature sensor string ends with "2"): # offset occurs, all sensor must starts from 1
    renumber_the_temperature_sensors()
```

We people doesn't care about the real cpu sensor name, haha.

exelban, 2020-12-08:
It's not so simple. I receive the list of keys from SMC. And after the mapping it with names. There is no ordering of sensors. 

There is one way to fix it. It's to change the mapping and processing of the keys which could be iterated (`TCxC`). It allows the mapping name to `x`.

closed 2020-12-13 by exelban