exelban/stats #216: Suspicion of system resources consumption forcing to reboot

bouscram, 2020-12-07:
**Describe the bug**

**_At the moment, I'm not sure that Stats is the cause, but things have been OK since I stopped it_**

I have seen since a few days (one week or a bit more), a strange phenomenon happening. At some point in time, it seems that OS resources are exhausted, which has several effects:

- Safari says it is missing essential parts and needs to be reinstalled
- External disks are not recognised anymore
- System is very slow and Stats' CPU widget shows constant activity (sometimes up to 100%)
- Apps are slow to start or can't even be launched

Memory consumption looks normal.

When this happens, the only thing possible is to reboot the machine.

I have stopped Stats since Saturday and things seem to be normal again.

**Details:**
- Device: end 2012 iMac
- MacOS: 10.15.7
- Application version: at least starting with 2.4.5
- Intel PowerGadget installed

exelban, 2020-12-07:
Hi. Thanks for the feedback.
It's really strange situation.

These points could not be related to the app: 
```
- Safari says it is missing essential parts and needs to be reinstalled
- External disks are not recognised anymore
```

Could you please check the CPU usage of Stats?
Will be great if you can post the top CPU processes from the popup view.

Maybe there's some leak, and it can be the reason for 100% CPU usage.

bouscram, 2020-12-07:
I must say I am very puzzled. CPU usage was generally higher than normal (about 25/35% when doing nothing) but I didn't see any unusual process in the list. Stats wasn't using more CPU that usually, i.e. less than 1%.

I guess that the Safari and disk problems were only side effects, like if a resource they need was not available. Memory seemed OK but I didn't check the size of the virtual memory used (I should have).

For the moment I have stopped Stats (at least for a few days, to be fairly sure that the issue was related). If I have some time, I'll do more testing and will keep you posted.

exelban, 2020-12-07:
There is an option that Stats consume a lot of CPU. If some leak happened. But I hope I catch all of them.

I'm very astonished by this behavior.
```
- Safari says it is missing essential parts and needs to be reinstalled
- External disks are not recognised anymore
```

This can not be related to CPU or Memory consumption. It looks like some problem with the system.

Because, for example, these two could be related to each other:
```
- System is very slow and Stats' CPU widget shows constant activity (sometimes up to 100%)
- Apps are slow to start or can't even be launched
```

bouscram, 2020-12-12:
Hi. After one week without Stats (where I didn't see any issue), I restarted this morning.

I immediately saw that the CPU was too high when the "Network" widget was shown. 
Then I found that at some point I had changed the working mode from the default of "Interface based" to "Process based", and that it makes a heavy use of the "nettop" program causing the CPU overload (up to 140% for that process only on my rather old 4-cores CPU).

I will leave Stats running to see what happens now.

exelban, 2020-12-12:
Hi. Thanks for the tests and feedback.

The difference between `Interface based` and `Process based` in the way that how the bandwidth stats collecting.
Interface based receive the data directly from the interface stats. Process based reads the output from the `nettop`. It's only one way how it's possible to receive the bandwidth per process.

exelban, 2020-12-20:
Hi. Any updates?

bouscram, 2020-12-20:
Hi. No problem since last week. If it was linked to Stats, it should be with its use of nettop (or nettop itself).

exelban, 2020-12-20:
Thanks. And even switching between `Interface based` and `Process based` does not helps?

bouscram, 2020-12-20:
I’ve only used the `interface based ` setting this week, which was the basis of my previous comment.

exelban, 2020-12-27:
Hi. I'm going to close this issue.

I take a look at this module. But don't see any option which could be optimized. If this bug related to the high CPU usage of `nettop` I cannot do anything with this.

Please reopen the issue if you will have some suggestions on how it could be fixed.

closed 2020-12-27 by exelban