exelban/stats #219: Feature request: switch temperature between F and C

extraPaul, 2020-12-08:
Because there is no indication of the units being used for temperature regarding the CPU and the GPU, I'm assuming it's currently displayed in Fahrenheit. It would be nice if there was an option to switch the temperature readings from Fahrenheit to Celsius.

wfrank21, 2020-12-09:
You actually can switch from Fahrenheit to Celsius or use your Macs default unit:
[image]

exelban, 2020-12-09:
Hi. 
As @wfrank21 write. It could be changed in the Sensors module. 
But I think this option must be moved to the application settings. Which must be separated from the System Information view. I have a plan to do it. But cannot find time for this.

closed 2020-12-13 by exelban

coreyk, 2020-12-14:
Choosing "System" from the Temperature settings shows Celsius even when OS is set to Fahrenheit:
[image]

exelban, 2020-12-14:
@coreyk thanks. Could you open a separate issue, please?