exelban/stats #221: Installing over brew doesn't work on Big Sur

theintz, 2020-12-09:
**Describe the bug**
Trying to install stats using brew on Big Sur just gives an error:
```
➜  ~ brew install --cask stats

Error: Cask 'stats' is unavailable: No Cask with this name exists.
```

exelban, 2020-12-09:
Hi. Thanks for reporting.
I try to install the app a few minutes ago with brew on Big Sur. And it was installed successfully.

I check the cask repo. And the Stats configuration [exists](https://github.com/Homebrew/homebrew-cask/blob/master/Casks/stats.rb). 

I don't know how exactly cask works. And cannot help with this problem(

theintz, 2020-12-09:
Hm, that's weird. Maybe my brew configuration is off? Installing with the dmg worked well.

exelban, 2020-12-09:
I'm not sure. But maybe in some way you have an old list of casks?
Do you try to install some other app with brew cask?

theintz, 2020-12-09:
When I copy the code `brew install --cask stats` from the website it doesn't work, when I type everythiing out it does. Very odd. Either way, it does work after all. Thanks

closed 2020-12-09 by theintz