exelban/stats #222: MacBook Air (M1, 2020) visual bugs

MrBenFTW, 2020-12-09:
**Describe the bug**
There are a few visual bugs I have noticed in the few hours I have been using stats for.

1. Battery health says 2% (good), I think this is due to the calculation not subtracting the degradation from 100. (100-2 = 98% health)
2. Only one of the CPU dials is filled, not sure what the other one is supposed to display. Also, I think it would be nice if we could display CPU core/thread usage for each core/thread
3. I also cannot see any method of checking system temperatures, yet in the battery view it is happy to report 30°c
4. On the topic of battery, I think it would be nice if we could see the current discharge in watts.

**Details:**
 - Device: `MacBook Air (M1, 2020)`
 - MacOS: `11.1 B2`
 - Application version: `2.4.7`
[image]
[image]
[image]

MrBenFTW, 2020-12-09:
By running in rosetta I am able to get the sensors to show up 
[image]
[image]
[image]

exelban, 2020-12-09:
Hi. Thanks for the feedback.

1. Nope, it's not substructing. It's a dividing `current capacity`/`designed capacity`. And it looks like `DesignCapacity` field is missing on m1. (need some help to research how it can be fixed).
2. In this circle must be the CPU temperature. But on Apple Silicon, there is no option to receive it (for now, see point nr.4). I will remove this circle if there is no temperature available. Till m1, this temperature always was available.
3. You could visualize the CPU usage per core/thread if you select `Bar Chart` widget.
4. On Apple Silicon there are no available sensors to show. At least at this moment. Please read [this](https://github.com/exelban/stats/issues/210) thread.
5. I'm just interested in why you need to have current discharge in watts? But yeah, it can be added.

MrBenFTW, 2020-12-09:
> 5. I'm just interested in why you need to have current discharge in watts? But yeah, it can be added.

It's a more standardised unit for measuring efficiency. It is also used to measure the TDP which is nice to compare against an x86 machine.

I am a developer myself so I might look into this project later and see if I can help you out, let me know how I can help you out with that research

exelban, 2020-12-09:
Just to summarize. There are 2 bugs which can be fixed:
- hide the temperature indicator if there is no temperature
- fix the battery health (there I need your help)

And show a current battery discharge in watts.

hexaeder, 2020-12-09:
Hey, thanks for the nice app! I have two additional informations from M1 mba:

> You could visualize the CPU usage per core/thread if you select Bar Chart widget.

Using the bar chart there is still only one process visible.
[image]

Besides that, the memory widget shows different data than the activity monitor. I couldn't manage to reproduce but the difference in app memory was several gb at some point.
[image]
[image]

exelban, 2020-12-09:
@hexaeder thanks. You could switch the toggle in the settings to show usage per core.
Now it looks like the RAM is the same.

exelban, 2020-12-09:
@MrBenFTW could you please post the output of `ioreg -l -n AppleSmartBattery -r | grep Capacity`?

MrBenFTW, 2020-12-09:
> ioreg -l -n AppleSmartBattery -r | grep Capacity

```
      "CurrentCapacity" = 68
      "AppleRawCurrentCapacity" = 2776
      "AbsoluteCapacity" = 0
      "BatteryData" = {"Ra03"=0,"Ra10"=0,"CellWom"=(0,0),"RaTableRaw"=(<005500520044004b005c0069004e005e00600064006b006d008300c30190027b>,<0000005a004b0051006300750073007d007200730077007a008b00c6019f0292>,<0055005f004f005a006d008100770080007d0078007e007f008f00d0019e0298>),"Qstart"=0,"AdapterPower"=969275305,"DailyMinSoc"=66,"Ra04"=0,"Ra11"=0,"CellVoltage"=(3945,3946,3942),"PackCurrentAccumulator"=18446744073705405607,"PassedCharge"=38,"Flags"=16777217,"PresentDOD"=(34,34,34),"Ra05"=0,"Ra12"=0,"FccComp1"=4272,"ChemID"=10037,"iMaxAndSocSmoothTable"=<0000000000000000000000000000000000000000000000000000000000000000>,"FccComp2"=4272,"PackCurrentAccumulatorCount"=25853,"DOD0"=(5488,5472,5536),"Ra06"=0,"ResScale"=0,"Ra13"=0,"WeightedRa"=0,"FilteredCurrent"=0,"RSS"=0,"CellCurrentAccumulatorCount"=0,"Serial"="F5D0375CKJVPJYV3B","DailyMaxSoc"=91,"Ra07"=0,"Ra14"=0,"MaxCapacity"=100,"ChemicalWeightedRa"=0,"Ra00"=0,"BatteryHealthMetric"=0,"DesignCapacity"=4382,"Ra08"=0,"BatteryState"=<000000000000000043e400020d000002>,"CellCurrentAccumulator"=(0,0),"AlgoChemID"=10037,"MfgData"=<00000000100200012735000003313731033030320341544c001b000000000000>,"ManufactureDate"=55186994247730,"ISS"=18446744073709551427,"Ra01"=0,"Soc1Voltage"=0,"QmaxDisqualificationReason"=0,"ChargeAccum"=0,"SimRate"=0,"Qmax"=(4536,4543,4486),"PMUConfigured"=0,"ITMiscStatus"=0,"StateOfCharge"=65,"Ra09"=0,"GaugeFlagRaw"=192,"CycleCount"=10,"Voltage"=11829,"SystemPower"=1074683891,"LifetimeData"={"Raw"=<0001d2ef0004958d00000f95000000000076a8490000e447b24000000000000001d5004b10fa0d3932de28260e36f532115df2fdf8aff70200e4000087010004>,"UpdateTime"=1607533253,"ResistanceUpdatedDisabledCount"=0,"TimeAtHighSoc"=<21000000530000000100000000000000>,"TemperatureSamples"=34561,"TotalOperatingTime"=2160,"MaximumDischargeCurrent"=18446744073709548850,"MinimumPackVoltage"=10278,"MaximumPackVoltage"=13022,"MaximumChargeCurrent"=3638,"AverageTemperature"=18446744073709551588,"MinimumTemperature"=75,"RDISCnt"=0,"MaximumTemperature"=469},"Ra02"=0}
      "NominalChargeCapacity" = 4382
      "MaxCapacity" = 100
      "DesignCapacity" = 4382
      "AppleRawMaxCapacity" = 4272
```

exelban, 2020-12-09:
@MrBenFTW thanks. It looks like you have a 2% battery capacity from designed)

Mr0cket, 2020-12-10:
@exelban It looks like the `CurrentCapacity` is using a different measure (%age?) in M1 as opposed to previous intel-based. 
these are the stats of my 2018 MPB: 
```
"AppleRawCurrentCapacity" = 4002
      "AppleRawMaxCapacity" = 4005
      "MaxCapacity" = 4005
      "CurrentCapacity" = 4002
      "LegacyBatteryInfo" = {"Amperage"=0,"Flags"=5,"Capacity"=4005,"Current"=4002,"Voltage"=12908,"Cycle Count"=498}
      "BatteryData" = {"LifetimeData"={"TotalOperatingTime"=99149874,"TimeAtHighSoc"=<1b9c0c00364a2c011642130000000000>,"UpdateTime"=1607598385},"Serial"="D867424B0EJH06FAM","ChemID"=13926,"Flags"=16608,"PassedCharge"=18446744073709551600,"Voltage"=12907,"ResScale"=200,"CycleCount"=498,"StateOfCharge"=100,"Qmax"=(4541,4652,4570),"DesignCapacity"=4790,"SystemPower"=10214,"PMUConfigured"=0,"DOD0"=(472,472,472),"PresentDOD"=(2,2,2),"CellVoltage"=(4303,4303,4302)}
      "DesignCapacity" = 4790
```

exelban, 2020-12-10:
It's strange. It cannot be the age. Because it's capacity)
Maybe someone else with m1 could check this values?

yokonrad, 2020-12-11:
I dont want to start a new issue, because there were started a subject about battery ;) I want to report some differences between battery module and coconutBattery. Im using the second one for many many years. I think their info about battery is real and i read a lot about that. But i dont know why there are differences in temperatures and percentages of battery. Sometimes there is same percentage and sometimes there is difference in 3-4%. Time until empty is also not the same. Why?

And if you can implement current discharge, would be awesome ;) Or maybe should i make a new issue?

[image]
[image]

lucasaf04, 2020-12-11:
@exelban  M1 MacBook Pro output:
```
"AppleRawCurrentCapacity" = 5038
"NominalChargeCapacity" = 5103
"AppleRawMaxCapacity" = 5170
"MaxCapacity" = 100
"CurrentCapacity" = 100
"BatteryData" = {"Ra03"=0,"Ra10"=0,"CellWom"=(0,0),"RaTableRaw"=(<0055007600640073007b00850067006a007000720074006f008400bb013b01ea>,<0055005d004e005a0062006b004f00520057005b0060005b006f00af014d0201>,<0055007600640073007d00880069006d0073007400780073008900c8015a0219>),"Qstart"=0,"AdapterPower"=1087542125,"DailyMinSoc"=98,"Ra04"=0,"Ra11"=0,"CellVoltage"=(4268,4268,4267),"PackCurrentAccumulator"=18446744073709551178,"PassedCharge"=0,"Flags"=553648641,"PresentDOD"=(4,4,4),"Ra05"=0,"Ra12"=0,"FccComp1"=5170,"ChemID"=9216,"iMaxAndSocSmoothTable"=<0000000000000000000000000000000000000000000000000000000000000000>,"FccComp2"=5170,"PackCurrentAccumulatorCount"=37,"DOD0"=(744,744,752),"Ra06"=0,"ResScale"=0,"Ra13"=0,"WeightedRa"=0,"FilteredCurrent"=0,"RSS"=0,"CellCurrentAccumulatorCount"=0,"Serial"="D860454A819L3L2EP","DailyMaxSoc"=98,"Ra07"=0,"Ra14"=0,"MaxCapacity"=100,"ChemicalWeightedRa"=0,"Ra00"=0,"BatteryHealthMetric"=0,"DesignCapacity"=5103,"Ra08"=0,"BatteryState"=<000000000000000043e4401085000002>,"CellCurrentAccumulator"=(0,0),"AlgoChemID"=9216,"MfgData"=<00000000100200012400000003313533033030330341544c0141000000000000>,"ManufactureDate"=57381588383794,"ISS"=0,"Ra01"=0,"Soc1Voltage"=0,"QmaxDisqualificationReason"=0,"ChargeAccum"=0,"SimRate"=0,"Qmax"=(5521,5497,5521),"PMUConfigured"=2904,"ITMiscStatus"=0,"StateOfCharge"=98,"Ra09"=0,"GaugeFlagRaw"=16608,"CycleCount"=2,"Voltage"=12801,"SystemPower"=1083699177,"LifetimeData"={"Raw"=<000000000015cefb00002ad600000000002f43c30000e447b24000000000000001a1005610ff0d9b32f0293610e8f3ca1492f2f4f580f35a00ef000035c70002>,"UpdateTime"=1607671485,"ResistanceUpdatedDisabledCount"=0,"TimeAtHighSoc"=<000000008d0100000300000000000000>,"TemperatureSamples"=13767,"TotalOperatingTime"=860,"MaximumDischargeCurrent"=18446744073709548490,"MinimumPackVoltage"=10550,"MaximumPackVoltage"=13040,"MaximumChargeCurrent"=4328,"AverageTemperature"=18446744073709551599,"MinimumTemperature"=86,"RDISCnt"=0,"MaximumTemperature"=417},"Ra02"=0}
"AbsoluteCapacity" = 0
"DesignCapacity" = 5103
```
Fully charged and the power source is the power adapter.
Hope this helps.

exelban, 2020-12-11:
@lucasaf04 thanks. Could you share the same stats but when the battery will be not full?

lucasaf04, 2020-12-11:
@exelban Here they are.
```
"CurrentCapacity" = 94
"AppleRawCurrentCapacity" = 4651
"AbsoluteCapacity" = 0
"BatteryData" = {"Ra03"=0,"Ra10"=0,"CellWom"=(0,0),"RaTableRaw"=(<0000007600640073007b00850067006a007000720074006f008400bb013b01ea>,<0000005d004e005a0062006b004f00520057005b0060005b006f00af014d0201>,<0000007600640073007d00880069006d0073007400780073008900c8015a0219>),"Qstart"=0,"AdapterPower"=961291357,"DailyMinSoc"=90,"Ra04"=0,"Ra11"=0,"CellVoltage"=(4143,4151,4141),"PackCurrentAccumulator"=18446744073708071664,"PassedCharge"=357,"Flags"=16777217,"PresentDOD"=(11,11,11),"Ra05"=0,"Ra12"=0,"FccComp1"=5204,"ChemID"=9216,"iMaxAndSocSmoothTable"=<0000000000000000000000000000000000000000000000000000000000000000>,"FccComp2"=5204,"PackCurrentAccumulatorCount"=29727,"DOD0"=(864,864,880),"Ra06"=0,"ResScale"=0,"Ra13"=0,"WeightedRa"=0,"FilteredCurrent"=0,"RSS"=0,"CellCurrentAccumulatorCount"=0,"Serial"="D860454A819L3L2EP","DailyMaxSoc"=98,"Ra07"=0,"Ra14"=0,"MaxCapacity"=100,"ChemicalWeightedRa"=0,"Ra00"=0,"BatteryHealthMetric"=0,"DesignCapacity"=5103,"Ra08"=0,"BatteryState"=<000000000000000043e4000284000002>,"CellCurrentAccumulator"=(0,0),"AlgoChemID"=9216,"MfgData"=<00000000100200012400000003313533033030330341544c0141000000000000>,"ManufactureDate"=57381588383794,"ISS"=18446744073709551356,"Ra01"=0,"Soc1Voltage"=0,"QmaxDisqualificationReason"=0,"ChargeAccum"=0,"SimRate"=0,"Qmax"=(5521,5497,5521),"PMUConfigured"=0,"ITMiscStatus"=0,"StateOfCharge"=90,"Ra09"=0,"GaugeFlagRaw"=192,"CycleCount"=2,"Voltage"=12418,"SystemPower"=1078663455,"LifetimeData"={"Raw"=<000000000015cefb00002ad600000000002f43c30000e447b24000000000000001a1005610ff0d9b32f0293610e8f3ca1492f2f4f580f35a00ef000035c70002>,"UpdateTime"=1607701174,"ResistanceUpdatedDisabledCount"=0,"TimeAtHighSoc"=<000000008d0100000300000000000000>,"TemperatureSamples"=13767,"TotalOperatingTime"=860,"MaximumDischargeCurrent"=18446744073709548490,"MinimumPackVoltage"=10550,"MaximumPackVoltage"=13040,"MaximumChargeCurrent"=4328,"AverageTemperature"=18446744073709551599,"MinimumTemperature"=86,"RDISCnt"=0,"MaximumTemperature"=417},"Ra02"=0}
"NominalChargeCapacity" = 5103
"MaxCapacity" = 100
"DesignCapacity" = 5103
"AppleRawMaxCapacity" = 5204
```
95% battery without power adapter.

exelban, 2020-12-11:
@lucasaf04 thanks. So, it looks like Capacity it's a battery level on m1(
I don't see the option to determine the battery health percentage based on these stats. Maybe someone could help with this?

exelban, 2020-12-11:
@yokonrad thanks. I could response here.

Stats is reading all battery statistics from `AppleSmartBattery `. It's an internal service which providing different params of the battery. I suppose default battery using this service too. I don't know how this app (`coconutBattery`) receiving the data. Maybe they calculating the time based on %. Maybe not. I don't know. And could not say why there is the difference between remaining times and percentage. Also it's a very strange that there is a difference in the percentage. Because it could not be 2 different values.

lucasaf04, 2020-12-11:
@exelban I think AppleRaw prefix (now) means that the output value is in mA and they have changed non-prefix Max/CurrentCapacity to battery percentage. Check @Mr0cket intel Mac photo for comparison: "prefix" and "non-prefix" values are equal.

exelban, 2020-12-11:
I'm not sure. I need 2 values: design capacity and current capacity.
Design capacity we have. So I just need current. `AppleRawCurrentCapacity` - looks more like the current battery level.
Maybe `NominalChargeCapacity`? But it's a little bit strange that this is equal to `DesignCapacity`.

lucasaf04, 2020-12-11:
The Mac is new, so it can be NominalChargeCapacity. Time will say.

exelban, 2020-12-13:
Hi.

I fixed all bugs except battery health percentage. I think it could be fixed a little bit later. When I will have more information about data which new macs provide about capacity.

I think this ticket could be reopened when the new info will be available.

closed 2020-12-13 by exelban

lasharor, 2021-08-15:
@exelban Is this fixed? Currently stats displays 100% on my MBA M1 whereas the MacOS settings panel displays 98%

exelban, 2021-08-16:
@lasharor Hi. Could you please share the output of `ioreg -l -n AppleSmartBattery -r | grep Capacity`?

lasharor, 2021-08-16:
> @lasharor Hi. Could you please share the output of `ioreg -l -n AppleSmartBattery -r | grep Capacity`?

  "CurrentCapacity" = 34
      "AppleRawCurrentCapacity" = 1332
      "AbsoluteCapacity" = 0
      "BatteryData" = {"Ra03"=0,"Ra10"=0,"CellWom"=(0,0),"RaTableRaw"=(<00000065005100480050007a00520063006a006e007700790077008d012401ce>,<00550063005a0051005a00830059006d006d0070007a0086008a00b301700242>,<00000076006600590066009000670076007c0078008c0089009100ba016e024b>),"Qstart"=0,"AdapterPower"=968616754,"DailyMinSoc"=34,"Ra04"=0,"Ra11"=0,"CellVoltage"=(3758,3754,3748),"PackCurrentAccumulator"=18446744073699472861,"PassedCharge"=1099,"Flags"=16777217,"PresentDOD"=(64,63,64),"Ra05"=0,"Ra12"=0,"FccComp1"=4060,"ChemID"=10037,"iMaxAndSocSmoothTable"=<0000000000000000000000000000000000000000000000000000000000000000>,"FccComp2"=4096,"PackCurrentAccumulatorCount"=64284,"DOD0"=(6400,6368,6448),"Ra06"=0,"ResScale"=0,"Ra13"=0,"WeightedRa"=0,"FilteredCurrent"=0,"RSS"=0,"CellCurrentAccumulatorCount"=0,"Serial"="D86044403NHPJYR38","DataFlashWriteCount"=2400,"DailyMaxSoc"=100,"Ra07"=0,"Ra14"=0,"MaxCapacity"=100,"ChemicalWeightedRa"=0,"Ra00"=0,"BatteryHealthMetric"=0,"DesignCapacity"=4382,"Ra08"=0,"BatteryState"=<000000000000000043e400024c000002>,"CellCurrentAccumulator"=(0,0),"AlgoChemID"=10037,"MfgData"=<00000000100200012735000003313731033030320341544c003b000000000000>,"ManufactureDate"=61788208052274,"ISS"=18446744073709551067,"Ra01"=0,"Soc1Voltage"=0,"QmaxDisqualificationReason"=0,"ChargeAccum"=0,"SimRate"=0,"Qmax"=(4337,4337,4332),"PMUConfigured"=0,"ITMiscStatus"=0,"StateOfCharge"=33,"Ra09"=0,"GaugeFlagRaw"=192,"CycleCount"=127,"Voltage"=11233,"SystemPower"=1084345240,"LifetimeData"={"Raw"=<00035449004c1cac0006476500000000018008140000e447124000000000000001c2008b10ff0d6532ee28440d71f5dd1001f47df95cf82100f50001b4f10005>,"UpdateTime"=1629102085,"ResistanceUpdatedDisabledCount"=0,"CycleCountLastQmax"=126,"TimeAtHighSoc"=<3c000000690500007200000000000000>,"TemperatureSamples"=111857,"TotalOperatingTime"=6991,"MaximumDischargeCurrent"=18446744073709549021,"MinimumPackVoltage"=10308,"MaximumPackVoltage"=13038,"MaximumChargeCurrent"=3441,"AverageTemperature"=18446744073709551605,"MinimumTemperature"=139,"RDISCnt"=0,"MaximumTemperature"=450},"Ra02"=0}
      "NominalChargeCapacity" = 4382
      "MaxCapacity" = 100
      "BatteryShutdownReason" = {"prevVoltage"=10884,"fullChargeCapacity"=3997,"UpoReboot"=0,"prevNominalChargeCapacity"=4382,"ResScale"=0,"RSS"=0,"MaximumRa0-8"=0,"TimeAbove95Perc"=22,"averageCurrent"=18446744073709551116,"prevFullChargeCapacity"=3997,"PassedCharge"=654,"uiSOC"=1,"PresentDOD"=95,"RaTableRaw"=<0000006600520051005a008200570069006f0076007e0081007e0095013601eb>,"prevAverageCurrent"=18446744073709551109,"remainingCapacity"=0,"prevRemainingCapacity"=0,"DOD0"=0,"NominalChargeCapacity"=4382,"timestamp"=1628834112,"MaximumDischargeCurrent"=18446744073709549021,"Voltage"=10871,"CycleCount"=124,"Qstart"=0,"Temperature"=3067,"DLog"=<a1fef8feebfe4afe97feeefef5fefafef9fef4fe4afe9afef8fef6fefcfef5fef0fe45fe78fd05fe0cfe5bfe8dfeeffef0feeefeb9fef5fe52fe38feeafed5fe>,"CycleCountLastQmax"=0,"DataError"=0}
      "DesignCapacity" = 4382
      "AppleRawMaxCapacity" = 4060

lasharor, 2021-08-16:
[image]
Really strange, this is the screen I get in settings.

MrBenFTW, 2021-08-16:
> Really strange, this is the screen I get in settings.

What does coconut battery say? I found that coconut is usually more accurate than macOS (it doesn't round)

lasharor, 2021-08-16:
> > Really strange, this is the screen I get in settings.
> 
> What does coconut battery say? I found that coconut is usually more accurate than macOS (it doesn't round)

[image]

MrBenFTW, 2021-08-16:
Here is mine BTW:
```
benmitchell@Scylla ~ % ioreg -l -n AppleSmartBattery -r | grep Capacity
      "CurrentCapacity" = 95
      "AppleRawCurrentCapacity" = 3611
      "AbsoluteCapacity" = 0
      "BatteryData" = {"Ra03"=0,"Ra10"=0,"CellWom"=(0,0),"RaTableRaw"=(<00000053004600430051007e004f005e0060006f00660059006700a401500217>,<005500700057004d00610098005f0067006e0078007800610073008c012d01db>,<00000074005900550066009a0061006b0072007e007c0068007400a100ca0147>),"Qstart"=0,"AdapterPower"=863182601,"DailyMinSoc"=91,"Ra04"=0,"Ra11"=0,"CellVoltage"=(4119,4116,4114),"PackCurrentAccumulator"=18446744073708341303,"PassedCharge"=18446744073709549405,"Flags"=16777217,"PresentDOD"=(12,12,12),"Ra05"=0,"Ra12"=0,"FccComp1"=4023,"ChemID"=10037,"iMaxAndSocSmoothTable"=<0000000000000000000000000000000000000000000000000000000000000000>,"FccComp2"=4059,"PackCurrentAccumulatorCount"=1952,"DOD0"=(10240,10304,10368),"Ra06"=0,"ResScale"=0,"Ra13"=0,"WeightedRa"=0,"FilteredCurrent"=0,"RSS"=0,"CellCurrentAccumulatorCount"=0,"Serial"="F5D04676AFUPJYVAT","DataFlashWriteCount"=2726,"DailyMaxSoc"=99,"Ra07"=0,"Ra14"=0,"MaxCapacity"=100,"ChemicalWeightedRa"=0,"Ra00"=0,"BatteryHealthMetric"=0,"DesignCapacity"=4382,"Ra08"=0,"BatteryState"=<000000000000000043e4000247000002>,"CellCurrentAccumulator"=(0,0),"AlgoChemID"=10037,"MfgData"=<00000000100200012735000003313731033030320341544c001b000000000000>,"ManufactureDate"=57381655427122,"ISS"=18446744073709551014,"Ra01"=0,"Soc1Voltage"=0,"QmaxDisqualificationReason"=0,"ChargeAccum"=0,"SimRate"=0,"Qmax"=(4336,4328,4310),"PMUConfigured"=0,"ITMiscStatus"=0,"StateOfCharge"=90,"Ra09"=0,"GaugeFlagRaw"=192,"CycleCount"=137,"Voltage"=12360,"SystemPower"=1089978385,"LifetimeData"={"Raw"=<000d40cb004a5ba50008ad290000066a016bae0c0000e447b24000000000000001d6003910fa0c4832e625ca0e7ef3d11144f25df83cf7b000ee00019dca0004>,"UpdateTime"=1629122749,"ResistanceUpdatedDisabledCount"=0,"CycleCountLastQmax"=137,"TimeAtHighSoc"=<f1000000490500009d00000000000000>,"TemperatureSamples"=105930,"TotalOperatingTime"=6620,"MaximumDischargeCurrent"=18446744073709548497,"MinimumPackVoltage"=9674,"MaximumPackVoltage"=13030,"MaximumChargeCurrent"=3710,"AverageTemperature"=18446744073709551598,"MinimumTemperature"=57,"RDISCnt"=0,"MaximumTemperature"=470},"Ra02"=0}
      "NominalChargeCapacity" = 4382
      "MaxCapacity" = 100
      "DesignCapacity" = 4382
      "AppleRawMaxCapacity" = 4023
benmitchell@Scylla ~ % 
```

MrBenFTW, 2021-08-16:
[image]
[image]
[image]

exelban, 2021-08-16:
@MrBenFTW Does coconutBattery show something like 91.8%?