exelban/stats #228: Only Upload speed is show, Download speed is Not showing

mhkanon, 2020-12-15:
Only Upload speed is show, Download speed is Not showing. See the attachment
[image]

Please, check and help to solve the problem.

Mehedi

exelban, 2020-12-15:
Hi.
Try to switch 'Reader type' to the `Processes based` in the Network settings (of Stats app).

Do you use some external wifi dongle?

mhkanon, 2020-12-15:
Thank you. Fixed the problem.

closed 2020-12-16 by exelban

mhkanon, 2020-12-22:
If the internet drops or electricity is gone speed show is hung(Internet is working but the apps show 0.0Kb/s). After shutting down the apps and reopen it works again.