exelban/stats #230: Inaccurate CPU readings

kim0, 2020-12-16:
**Describe the bug**
The attached screenshot shows `WindowServer` is consuming 18%, while the total is showing as only 14%. So something is off in the CPU consumption calculations.

**Details:**
 - Device: Macbook Pro 2013
 - MacOS 11.0.1
 - Application version 2.4.8

[image]

exelban, 2020-12-16:
Duplicate of #115

closed 2020-12-16 by exelban

kim0, 2020-12-16:
Thanks @exelban .. But I am not comparing against Activity Monitor, or another tool. This app "stats" gives a breakdown of readings, which when added up, are more than the show total. I'm guessing that's a different issue ?

exelban, 2020-12-16:
Process usage has Unix style.
Total usage has a "0 - 100%" style.