exelban/stats #231: Incorrect temperature for dGPU, MacBook Pro 16 2019

richardtop, 2020-12-18:
**Describe the bug**
Stats displays 0 C for the dGPU temperature

[image]

**Details:**
 - Device: MacBook Pro 16 2019
 - MacOS 11.1
 - Application version 2.4.8

exelban, 2020-12-18:
Hi. It means that the temperature for this GPU is unavailable.

richardtop, 2020-12-18:
OK, I suggest hiding the chart and the value completely, would that be possible?

exelban, 2020-12-18:
Yeap, I think it's possible.

Military-Doctor, 2020-12-28:
I think this problem was fixed in commit [a59ecf4](https://github.com/exelban/stats/pull/245/commits/a59ecf45122e566e177734087486e4d09e5b3bcb).

exelban, 2020-12-28:
@richardtop please recheck if now you can see the AMD temperature.

richardtop, 2020-12-28:
Fixed.

closed 2020-12-28 by richardtop