exelban/stats #233: Disk menu showing (wrong?) data

andreialecu, 2020-12-18:
**Describe the bug**

First of all, thanks for this amazing app!

I'm attaching comparison screenshots between the built-in About this Mac output and what Stats displays:

[image]

[image]

I'm using an M1 Mac Mini and I noticed that the "Disk Usage" seems to be reported wrong.

The disk size is 228.27 vs 245.11 on one vs the other. Also the space available differs.

It's possible I'm misunderstanding what it is supposed to show, but it seems the information in the "About This Mac" app is more useful. Is this a bug or by design?

**Details:**
 - Device: M1 Mac Mini
 - MacOS 11.1
 - Application version 2.4.8

exelban, 2020-12-18:
Hi. Thanks for reporting.
It looks like the bug in the partition detection by Stats. I will try to debug it.

exelban, 2020-12-18:
Could you please post the output of `diskutil apfs list`?

qkdxorjs1002, 2020-12-19:
**Details:**

- Device: MBP 13' 2020 4ports
- MacOS 11.1
- Application version 2.4.8

[image]
[image]

`❯ diskutil apfs list
APFS Container (1 found)
|
+-- Container disk1 8E46163A-A90A-4045-A221-E04B509F6B39
    ====================================================
    APFS Container Reference:     disk1
    Size (Capacity Ceiling):      499963174912 B (500.0 GB)
    Capacity In Use By Volumes:   97636003840 B (97.6 GB) (19.5% used)
    Capacity Not Allocated:       402327171072 B (402.3 GB) (80.5% free)
    |
    +-< Physical Store disk0s2 DFFB3654-267F-4E1A-8F62-113E8775B6A6
    |   -----------------------------------------------------------
    |   APFS Physical Store Disk:   disk0s2
    |   Size:                       499963174912 B (500.0 GB)
    |
    +-> Volume disk1s1 78C68721-147C-467A-8A19-D741A48C2DDF
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s1 (Data)
    |   Name:                      Macintosh HD - 데이터 (Case-insensitive)
    |   Mount Point:               /System/Volumes/Data
    |   Capacity Consumed:         77125947392 B (77.1 GB)
    |   Sealed:                    No
    |   FileVault:                 No (Encrypted at rest)
    |
    +-> Volume disk1s2 9483124D-13E5-4AD8-8927-75C474FA566C
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s2 (System)
    |   Name:                      Macintosh HD (Case-insensitive)
    |   Mount Point:               Not Mounted
    |   Capacity Consumed:         15046447104 B (15.0 GB)
    |   Sealed:                    Broken
    |   FileVault:                 No (Encrypted at rest)
    |   |
    |   Snapshot:                  BA2A6D75-B66C-42F2-ACC7-14530565CDF8
    |   Snapshot Disk:             disk1s2s1
    |   Snapshot Mount Point:      /
    |   Snapshot Sealed:           Yes
    |
    +-> Volume disk1s3 30422B87-2967-461B-9B1B-BD0BCF0D4C42
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s3 (Preboot)
    |   Name:                      Preboot (Case-insensitive)
    |   Mount Point:               /System/Volumes/Preboot
    |   Capacity Consumed:         326549504 B (326.5 MB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s4 2F1DCFCA-A25F-4574-AAE1-8FED423226F6
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s4 (Recovery)
    |   Name:                      Recovery (Case-insensitive)
    |   Mount Point:               Not Mounted
    |   Capacity Consumed:         613629952 B (613.6 MB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s5 38A151DC-947E-46BF-A369-B387701C2FEE
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s5 (VM)
    |   Name:                      VM (Case-insensitive)
    |   Mount Point:               /System/Volumes/VM
    |   Capacity Consumed:         20480 B (20.5 KB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s6 2EDF0A5A-6E7A-4BAD-B28B-069D79EB33AE
        ---------------------------------------------------
        APFS Volume Disk (Role):   disk1s6 (No specific role)
        Name:                      Backup (Case-insensitive)
        Mount Point:               /Volumes/Backup
        Capacity Consumed:         4361064448 B (4.4 GB)
        Sealed:                    No
        FileVault:                 No (Encrypted at rest)
`

andreialecu, 2020-12-19:
```
➜ diskutil apfs list                      
APFS Containers (3 found)
|
+-- Container disk3 B4710FE0-BCB8-4004-8AA3-8DD27339BAC0
    ====================================================
    APFS Container Reference:     disk3
    Size (Capacity Ceiling):      245107195904 B (245.1 GB)
    Capacity In Use By Volumes:   212160872448 B (212.2 GB) (86.6% used)
    Capacity Not Allocated:       32946323456 B (32.9 GB) (13.4% free)
    |
    +-< Physical Store disk0s2 41FA0E98-50C3-4D3C-A53B-3BFDAE165ABE
    |   -----------------------------------------------------------
    |   APFS Physical Store Disk:   disk0s2
    |   Size:                       245107195904 B (245.1 GB)
    |
    +-> Volume disk3s1 795A24EF-2B5C-47DF-9CC0-67661107CFE7
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk3s1 (System)
    |   Name:                      Macintosh HD (Case-insensitive)
    |   Mount Point:               Not Mounted
    |   Capacity Consumed:         15046959104 B (15.0 GB)
    |   Sealed:                    Broken
    |   FileVault:                 No (Encrypted at rest)
    |   |
    |   Snapshot:                  D4722794-A126-4384-86FA-FEAF8D5FF518
    |   Snapshot Disk:             disk3s1s1
    |   Snapshot Mount Point:      /
    |   Snapshot Sealed:           Yes
    |
    +-> Volume disk3s2 354DB073-DEB7-4B89-8A6E-9D90B293B4ED
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk3s2 (Preboot)
    |   Name:                      Preboot (Case-insensitive)
    |   Mount Point:               /System/Volumes/Preboot
    |   Capacity Consumed:         324419584 B (324.4 MB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk3s3 2A087EB7-1863-425F-A8B5-ACF15D413513
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk3s3 (Recovery)
    |   Name:                      Recovery (Case-insensitive)
    |   Mount Point:               Not Mounted
    |   Capacity Consumed:         1032253440 B (1.0 GB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk3s5 3E8F8945-B54E-4FAC-90E1-EDE9FB0C917C
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk3s5 (Data)
    |   Name:                      Data (Case-insensitive)
    |   Mount Point:               /System/Volumes/Data
    |   Capacity Consumed:         180585209856 B (180.6 GB)
    |   Sealed:                    No
    |   FileVault:                 No (Encrypted at rest)
    |
    +-> Volume disk3s6 A121619D-6476-486C-AA46-44B98F73E8C2
        ---------------------------------------------------
        APFS Volume Disk (Role):   disk3s6 (VM)
        Name:                      VM (Case-insensitive)
        Mount Point:               /System/Volumes/VM
        Capacity Consumed:         15032524800 B (15.0 GB)
        Sealed:                    No
        FileVault:                 No
```

exelban, 2020-12-19:
Hmm, thanks. It's super strange. Could I ask you to run a debug version and post the logs?

andreialecu, 2020-12-19:
Sure, let me know what the instructions are and I'll try to run it tomorrow and post the logs.

exelban, 2020-12-19:
Ok, I figured out where is the problem. macOS returns the size of the disk not in bytes. For example, I have a ~250GB disk, and macOS returns `249292435456`. I suppose that it will be bytes. And convert it's as bytes. In result `249292435456 B` = `232.17 GB`. I must change the calculation mechanism in this place.

closed 2020-12-20 by exelban

andreialecu, 2020-12-20:
Hey @exelban, I just installed the latest version but the space available still seems wrong:

[image]

[image]

exelban, 2020-12-20:
Yeap. But according to the data which you post:
```
APFS Container Reference:     disk3
    Size (Capacity Ceiling):      245107195904 B (245.1 GB)
    Capacity In Use By Volumes:   212160872448 B (212.2 GB) (86.6% used)
    Capacity Not Allocated:       32946323456 B (32.9 GB) (13.4% free)
```

The system reports that you have used `212.2 GB` and `32.9 GB`.

andreialecu, 2020-12-20:
Correct, I wonder where the About this Mac data is taken from or why it shows this difference. 🤔 

Perhaps it's space that can be reclaimed?

Disk Utility seems to display the same data as Stats.

[image]

andreialecu, 2020-12-20:
Yes, that seems to be it. See this screenshot from Disk Utility:

[image]

Notice the `Available` field.

andreialecu, 2020-12-20:
Perhaps relevant: https://developer.apple.com/forums/thread/127736

exelban, 2020-12-20:
Unfortunately, the data you have posted is only one way to receive the drive spacing. Without root access.
It could be some bug in macOS which will be fixed in the future. Or maybe it's related to m1. Hard to say.

andreialecu, 2020-12-20:
The discussion from the developer forums mentions this:

https://developer.apple.com/documentation/foundation/nsurlvolumeavailablecapacityforimportantusagekey

Does it require root access? Seems it was posted 11 months ago, before M1 existed. And I also see similar discussions related to "purgeable" space in various other forums from before that as well. Seems like it should be very common on MacOS.

You don't see differences on your own Mac between the output of Stats and Disk Utility/About this Mac?

exelban, 2020-12-20:
Nope. On all my macs I have correct spacing.

exelban, 2020-12-20:
Very strange. If even Disk Utility and `diskutil` shows the wrong spacing. I suppose the problem is on another level.

exelban, 2020-12-20:
Try this one:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5720956/Stats.dmg.zip)

andreialecu, 2020-12-20:
Here's a screenshot from an Intel-based 2018 MacBook Pro, so it looks like the "purgeable space" difference exists on every Mac:

[image]

andreialecu, 2020-12-20:
> Try this one:
> [Stats.dmg.zip](https://github.com/exelban/stats/files/5720956/Stats.dmg.zip)

I think this one calculates it properly! 🚀 

[image]

`245.11 - 155.97 = 89.14` which is exactly the same as I see in the About this Mac.

exelban, 2020-12-20:
Ok. Thanks for your help.
I will release this fix next week. Till then you can use this build.