exelban/stats #236: Process list in RAM tab contains truncated names

richardtop, 2020-12-20:
**Describe the bug**
Process list in RAM tab contains truncated names, as indicated on the screenshot. Other tabs work just fine.

## RAM tab
[image]

## CPU tab
[image]

**Details:**
 - Device: Macbook Pro 2019
 - MacOS 11.1
 - Application version 2.4.8

exelban, 2020-12-20:
Hi. Thanks for reporing.
I will take a look at this bug.

exelban, 2020-12-27:
Hi. 
Unfortunately, I can't fix this.

I use `top` to receive the list of apps. And it has a limit to the process command. And `Google Chrome Helper` does not fit into it.
In the CPU module, it shows the proper command because it uses `ps` not `top`. And `ps` does not have such limitations.

closed 2020-12-27 by exelban