exelban/stats #237: Duplicate terms in Localizable.strings

Tai-Zhou, 2020-12-21:
I'm trying to update the zh-Hans translation but find two terms confusing.
https://github.com/exelban/stats/blob/8d3172594aa60e7f5818757c66526a5f3e6d2f0f/Stats/Supporting%20Files/en.lproj/Localizable.strings#L133
https://github.com/exelban/stats/blob/8d3172594aa60e7f5818757c66526a5f3e6d2f0f/Stats/Supporting%20Files/en.lproj/Localizable.strings#L145-L146
There are two "Interface" and I think one should be deleted. I also didn't find `LocalizedString("Process")` in the code so I can not decide the translation. Maybe these two terms are prepared for the "Reader type"?

exelban, 2020-12-21:
Hi. I suppose one of them could be removed.

Tai-Zhou, 2020-12-21:
Thank you! I think this issue can be closed.

closed 2020-12-22 by exelban