exelban/stats #238: 作者您好，我想请问下中文版里面关于"工序数"是什么意思。

947612227, 2020-12-21:
[image]

这是我电脑的配置
[image]

CPU有8个核心，16个线程，希望可以讲解下图一中圈出来的地方，什么是"工序数"

exelban, 2020-12-21:
Hi. Could you use English please?

Tai-Zhou, 2020-12-21:
> CPU有8个核心，16个线程，希望可以讲解下图一中圈出来的地方，什么是"工序数"

https://github.com/exelban/stats/blob/8d3172594aa60e7f5818757c66526a5f3e6d2f0f/Stats/Supporting%20Files/zh-Hans.lproj/Localizable.strings#L86

“工序数”是指在对应的弹窗中显示的“高占用进程数”。这个翻译是作者采用的，我猜应该是用的翻译软件。在 https://github.com/exelban/stats/issues/237 这个issue解决掉后我会提交一个新的翻译版本。

“工序数” corresponds to "Number of top processes" in the popup window. This translation is adopted by the author which I guess comes from a translation tool. I will submit a pull request of new translation after the issue https://github.com/exelban/stats/issues/237 is closed.

exelban, 2020-12-21:
Please use the English language.

Tai-Zhou, 2020-12-21:
> Please use the English language.

Sorry for inconvenience, in the last comment I just expressed the same meaning in both Chinese and English 😀, Chinese for @947612227 and English for you. Below is the translation of his initial comment https://github.com/exelban/stats/issues/238#issue-771964461:

---

Title: 作者您好，我想请问下中文版里面关于"工序数"是什么意思。

Hello dear author, I'm wondering about the meaning of "工序数" (Number of top processes) in the Chinese version.

> 这是我电脑的配置

This is the spec of my Mac.

> CPU有8个核心，16个线程，希望可以讲解下图一中圈出来的地方，什么是"工序数"

My CPU has 8 cores, 16 threads, I want to know what is "工序数" (Number of top processes) circled in the figure.

exelban, 2020-12-21:
Thanks. I suppose in the previous comment you give the answer to the question.
I'm going to close this issue.

closed 2020-12-21 by exelban