exelban/stats #241: GPU Auslastung auf M1 funktioniert nicht

dg0ojo, 2020-12-22:
Die Anzeige für die GPU Auslastung liegt immer bei 0% auf einem M1 Mac Mini

exelban, 2020-12-22:
Hi. Please use English.

closed 2020-12-23 by exelban

grandtom, 2020-12-24:
Hi @exelban,

I think that @dg0ojo wants to say that GPU always displays 0% on Apple M1. We can't change GPU to display.

[image]
[image]
[image]