exelban/stats #243: RAM/Memory Pie Chart Displaying Inactive as Free

rozhl, 2020-12-22:
When my RAM is 80 % full, the icon displays 50 % free. Also I would welcome a color wheel that just shows Used (Wired+Active+Inactive) as 1 part of the pie and Free as the other part. Ideally the color of the wheel would be user configurable. So I would have red/orange for used, white for free memory and a gray circle around the icon.

exelban, 2020-12-23:
Hi. 
Does it show the different values in the menu bar and in the popup?

I'm not sure that I could merge the values into one. In theory, it could be some toggle. To turn on a `sum mode`.

rozhl, 2020-12-24:
Hi,

The values are the same AFAIK, however the value about **Inactive memory is missing** altogether. For me it is important, because in my workflow when I start using disk cache, it slows me incredibly down.

Yes, a toggle would do.

Thanks!

exelban, 2020-12-24:
I don't understand the problem.
`When my RAM is 80 % full, the icon displays 50 % free.` - does it means that Stats shows the wrong values? And does the values in the popup view and in the menu bar the same?

andreialecu, 2020-12-27:
Seems that Stats is displaying wrong values on my M1 Mac Mini for the memory tab too. Comparison screenshots for Stats vs Activity Monitor:

[image]

[image]

The values shown are off by a big amount. Especially notice `swap` which is 16.78GB in Activity Monitor and just 988MB in Stats. Using Stats 2.4.9

exelban, 2020-12-27:
@andreialecu how it's related to this issue?

andreialecu, 2020-12-28:
@exelban to quote the first sentence of the OP:
> When my RAM is 80 % full, the icon displays 50 % free.

The screenshots I attached show the same issue. The values displayed by Stats do not match what Activity Monitor shows.

exelban, 2020-12-28:
@andreialecu In your case stats show the wrong value. The ticket is related to the Pie Chart visualization.

rozhl, 2020-12-28:
@exelban It's the same issue: @andreialecu 's used memory is 7 GB, total is 8 GB and Stats is showing 3 GB free in the pie chart and in the menu. So the pie chart is correctly showing the values, but the values are off by a big margin.

exelban, 2020-12-28:
@rozhl do you have m1?

rozhl, 2020-12-28:
@exelban No, Intel.

rozhl, 2020-12-28:
@exelban I downloaded iStat Menus demo and it is showing the correct values: (Stats on the left, iStat Menus on the right)
[image]

exelban, 2020-12-28:
Could you check [this](https://github.com/exelban/stats/releases/tag/v2.4.7) version?

exelban, 2020-12-28:
Oh wait. It was for apple silicon ready. Try [this](https://github.com/exelban/stats/releases/tag/v2.4.4) one please.

rozhl, 2020-12-28:
> Could you check [this](https://github.com/exelban/stats/releases/tag/v2.4.7) version?

The values are the same. I don't see a difference between the current and this one. The inactive memory is missing in both version.

rozhl, 2020-12-28:
> Oh wait. It was for apple silicon ready. Try [this](https://github.com/exelban/stats/releases/tag/v2.4.4) one please.

Shows 13 GB free when I have only 6 GB free. So the same bug in my eyes.

exelban, 2020-12-28:
Hmm. It looks like something was changed in macOS.

exelban, 2020-12-28:
I will try to fix it today.

exelban, 2020-12-28:
It looks like `application memory` is wrong.

andreialecu, 2020-12-28:
@exelban I can also confirm that `Wired` and `Compressed` matches Activity Monitor, but `App Memory` and `Swap` are both wrong on Stats.

rozhl, 2020-12-28:
> Hmm. It looks like something was changed in macOS.

On December 16th I installed the latest Security update and some apps had problems working until I reseted the SMC and NVRAM. Maybe there was something that changed...

> @exelban I can also confirm that `Wired` and `Compressed` matches Activity Monitor, but `App Memory` and `Swap` are both wrong on Stats.

Originally I noticed the memory used as "Inactive" not being reported and causing wrong "Free Memory" information.

exelban, 2020-12-28:
Swap is not really related to the RAM itself. Yeap, I can confirm too that after the last update it broke `active` (`App`) memory(

exelban, 2020-12-28:
@rozhl @andreialecu could you check this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5747710/Stats.dmg.zip)

andreialecu, 2020-12-28:
@exelban attaching screenshot from new build, side by side with Activity Monitor.

[image]

Seems a bit better, but still not exactly the same output as Activity Monitor. Swap is still wrong.

exelban, 2020-12-28:
Thanks. I totally don't care about the swap.

Very, very strange. According to the screenshot:
- app = 3 GB
- wired = 1.67 GB
- compressed = 1.23 GB
Summ = 5.9 GB.

But it shows 6.49 GB. Something is wrong here.

andreialecu, 2020-12-28:
Possibly relevant:

[image]

Adding up (via `vm_stat`):

- Pages Active: `164556`
- Pages wired down: `119876`
- Pages occupied by compressor: `34578`
- File-backed pages: `98593`

`* 16384`

Results in ~6.39GB which is close to what Activity Monitor shows.

exelban, 2020-12-28:
I'm not sure what `File-backed` means. Maybe you know?

exelban, 2020-12-28:
In the last build, I used this way to calculate used memory: `active + inactive + speculative + wired + compressed - purgeable - external`

rozhl, 2020-12-28:
> @rozhl @andreialecu could you check this build?
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/5747710/Stats.dmg.zip)

Still not correct... Attaching a screenshot comparing this Stats version with the iStat Menus roughly at the same time.

[image]

exelban, 2020-12-28:
@rozhl could you please compare to the Activity Monitor?

rozhl, 2020-12-28:
> @rozhl @andreialecu could you check this build?
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/5747710/Stats.dmg.zip)

There you go... the test version and the Activity Monitor at the same time:
[image]

exelban, 2020-12-28:
@rozhl according to this screenshot the values are right. But the circle showing the wrong data.

rozhl, 2020-12-28:
Yes. And there should also be an option to include Inactive Memory in the circle, since it doesn't matter what shows up here, the Inactive Memory is also used...

exelban, 2020-12-29:
Could you check this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5750299/Stats.dmg.zip)

andreialecu, 2020-12-29:
Here's a screenshot from the latest build. 

Swap seems fixed now, which is great because it seems pretty important to see especially on lower mem machines like the 8GB M1. 🎊 

However, `Used Memory` still seems wrong by about 10%. 5.80 GB vs 6.39 GB

andreialecu, 2020-12-29:
It seems that the calculation is probably incorrect because of Compressed memory somehow. @rozhl's screenshots don't have any compressed memory visible. It probably factors in to total usage somehow differently.

andreialecu, 2020-12-29:
> I'm not sure what `File-backed` means. Maybe you know?

I believe this is the `Cached Files` field in Activity Monitor. It seems to correlate with it.

andreialecu, 2020-12-29:
I just had a possible revelation. 

The M1 shares the system memory with the GPU as well, via the `unified memory` architecture. I guess it's possible that the missing RAM goes to the GPU and is not part of any paged memory and must be read from elsewhere?

exelban, 2020-12-29:
I have no idea why there is a difference. Even in the `Activity Monitor` there are some differences. Used memory must be equal to the: app + wired + compressed.
And `Physical Memory` it's: used + cached.

I'm pretty sure that even intel GPU vram using ram. I think apple GPU works in the same way.

rozhl, 2020-12-29:
> Could you check this build?
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/5750299/Stats.dmg.zip)

Sending the screenshot from this version...
[image]

exelban, 2020-12-29:
Thanks. Now it looks the same.
I'm just confused a little bit. How you have 24GB? Some iMac with 16+8 GB?

rozhl, 2020-12-29:
Yes, iMac with 16+8 GB.

rozhl, 2020-12-29:
Could you now also add the inactive memory to the calculation? As an option, if not by default, since that one affects me particularly:
[image]
Stats says 11 GB free when I have only 3 GB in fact...

exelban, 2020-12-29:
According to the `Activity Monitor` you have 11GB free, not 3GB.

Could you maybe explain why you need `Inactive` to be visualized?

exelban, 2020-12-29:
What I know about `Inactive` memory. It's a temporary space in the RAM where macOS put some data which could be useful in the future. For example, you open some App then close it. The app will not instantly free from RAM, it will be moved to `Inactive`. It's some `prediction` mechanism. Which could speed up the next App opening. When macOS need more RAM than have in `free` space, it will release some space from `inactive` memory.

TL.DR: `inactive` = `free`.

I think the better explanation you could find [here](https://discussions.apple.com/thread/4594176?answerId=4594176021#4594176021).

rozhl, 2020-12-29:
> According to the `Activity Monitor` you have 11GB free, not 3GB.
> 
> Could you maybe explain why you need `Inactive` to be visualized?

Yeah, in my workflow the system doesn't free up the Inactive memory when the RAM fills up and starts using the disk cache, which is an incredible slow down for me. So I need to keep an eye on it too. That way I can prevent the slow down in ways like closing some apps or windows. Theoretically I understand that Inactive should be treated as free, but it is not a reality for me. You can see I have turned on the option to include Inactive in the pie chart for Used memory in the iStat Menus...

exelban, 2020-12-29:
Hmm, it makes sense. I must think about how it could be implemented.

rozhl, 2020-12-29:
Maybe something like this:
[image]

exelban, 2020-12-29:
Yes, but if I adding inactive memory to the list, App field must be replaced with Active. Otherwise, it will not calculate.

exelban, 2020-12-29:
I cannot just add the `Inactive` field.

closed 2021-01-03 by exelban