exelban/stats #247: If the internet drops or electricity is gone speed show is hung(Internet is working but the apps show 0.0Kb/s). After shutting down the apps and reopen it works again.

mhkanon, 2020-12-24:
If the internet drops or electricity is gone speed show is hung(Internet is working but the apps show 0.0Kb/s). After shutting down the apps and reopen it works again.

_Originally posted by @mhkanon in https://github.com/exelban/stats/issues/228#issuecomment-749386964_

exelban, 2020-12-24:
Hi. Thanks for reporting.
I will take a look at this.

exelban, 2020-12-24:
Could you maybe specify what type of reader do you use. And what mac do you have?

mhkanon, 2020-12-24:
[image]
[image]

I use Hackintosh OC. USB Wifi card "Chipset: Realtek RTL8811AU"

I notice that last some days when electricity gone my wifi router shut down. When electricity back the route on and wifi working fine but Internet speed is not showing. It shows 0.0 Kb/s. The problem happens if I disconnect the device and reconnect sometimes. This time if I change the "reader type" process-based  =>  Interface based. It only shows upload speed.

exelban, 2020-12-24:
Does it happen after sleep also?

mhkanon, 2020-12-24:
> Does it happen after sleep also?

Yes

exelban, 2020-12-24:
Could I ask you to check this command `nettop -P -L 1 -k time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch`. And post the output if you could. Or just say it's empty or has all processes on 0.
Run it when this bug appears again. It will give me the information about the problem.

mhkanon, 2020-12-25:
[image]

exelban, 2020-12-25:
Thanks. Could you run `nettop -P -k time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch`. And check if the values changing when using the internet (when Stats shows 0/0).

mhkanon, 2020-12-26:
[image]
Sometime change

exelban, 2020-12-27:
Hi. Could I ask you to run this build and simulate this bug?
This build will write some logs to the file `log.txt` which you can find in `Documents`.

Could you post log file after?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5744795/Stats.dmg.zip)

mhkanon, 2020-12-27:
The problem happens now. can you please come to team viewer to live to see?

mhkanon, 2020-12-27:
[log.txt](https://github.com/exelban/stats/files/5744924/log.txt)

exelban, 2020-12-27:
Does the internet works but stats show 0/0 when you close the app?

mhkanon, 2020-12-28:
> Does the internet works but stats show 0/0 when you close the app?

Yes. Internet working but stats show 0/0.

exelban, 2020-12-29:
Thanks. It looks like the process reader returns 0/0. 

When the `process-based` option is selected. It calculating bandwidth from processes. Which fetching from `nettop `. According to the logs, this function (which read from `nettop`) returns 0/0. It means that nettop does not change the values. Or maybe there is some bug)

I can build an additional build where log the output from `nettop`. It will help to detect the place where 0/0 appears. Would you like to run a few more builds and share the logs?)

mhkanon, 2020-12-29:
'Would you like to run a few more builds and share the logs?"
Yes, No problem.

exelban, 2020-12-29:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5751522/Stats.dmg.zip)

mhkanon, 2020-12-30:
[log.txt](https://github.com/exelban/stats/files/5752881/log.txt)

mhkanon, 2020-12-30:
[log.txt](https://github.com/exelban/stats/files/5752909/log.txt)

exelban, 2021-01-01:
Hi. Sorry for the long response.
I take a look at logs. It looks like the app can't run `nettop`. `nettop` process throw error. It could be related to some permission inside macOS. Or with `nettop` itself.

I don't see an option on how it can be fixed from the app level.

mhkanon, 2021-01-02:
No way to solve it?

exelban, 2021-01-02:
Nope. I believe there is some problem related to the USB wifi. There were a lot of reports that USB dongles do not report the stats at all. Additionally in Hackintosh. There could be a lot of different reasons why it does not work properly.

closed 2021-01-03 by exelban

mhkanon, 2021-01-03:
I using this wifi software https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter

exelban, 2021-01-16:
@mhkanon please try to run `nettop` when stats stops working. And give to know if there is some error, or it shows something.

mhkanon, 2021-01-17:
[image]

exelban, 2021-01-17:
Thanks. It's when the `Stats` does not show the network stats?

mhkanon, 2021-01-17:
[image]
Yes "Stats" show 0.0/kb

exelban, 2021-01-17:
Hmm, but top processes works?

mhkanon, 2021-01-17:
No. Everything is freezing. This time If I stop states and restart it work again.

exelban, 2021-01-17:
The problem: 
I know where the problem appears. It appears when run `nettop` to read the processes. And run the process throws an error. 

Why? I don't know. 
How it can be fixed? I don't know.
It could be fixed? Yes, but only when I reproduce this bug. Or someone who has this bug can debug this behavior.

mhkanon, 2021-01-17:
After freezing it auto-fix within 15-20 min. Then work fine 5-10 min. Again freezing. Maximum time freeze for not using the internet.