exelban/stats #249: First letter of some processes missing in RAM widget

adamddh, 2020-12-24:
The first letter of only some processes seems to be missing when the RAM widget is opened. However, this problem goes away when any window is opened (the problem only occurs when I have no other applications running). It seems to be processes that don't have a symbol associated with them. 

Screenshots included. 
[image]
[image]

exelban, 2020-12-24:
Hi. Thanks for reporting.
This issue is related to #236.

exelban, 2020-12-27:
Hi. I think it must be fixed in the [new](https://github.com/exelban/stats/releases/tag/v2.4.10) release. Give me to know if nothing changed.

adamddh, 2020-12-27:
I updated the app, tried to reproduce the error and could not do so, so it looks like the issue has been resolved.

closed 2020-12-27 by adamddh