exelban/stats #250: Feature request: add battery usage graph.

Military-Doctor, 2020-12-26:
Just like the iPhone Battery page.

[image]

exelban, 2020-12-26:
Hi.

Yeap. I think it could be added.

ppizarror, 2021-04-02:
Could it be a solution if the battery percentage + charging status is saved to a file each X time? Then display a graph using the latest information... I've searched and found no other way to get the battery usage using a terminal command

exelban, 2021-05-15:
Hi. I take a look at this feature. And there a few problems with it. First of all: the app does not have a battery level when mac in sleep mode. So there will be missing history. The other one is that I'm trying to not save any data (even locally). So after app restart, battery history will be empty.

And I'm not sure that this feature will be nice with these two "bugs".

exelban, 2021-05-23:
Hi again. I'm going to close this proposal. Please give to know if someone will have an idea how the data losing in sleep could be fixed. Then this feature could be implemented.

closed 2021-05-23 by exelban