exelban/stats #251: Feature request: show primary GPU in the menu bar.

Military-Doctor, 2020-12-26:
Thanks for the nice app!

I am currently using MacBook Pro 16" 2019, always carry my laptop without AC adapters. 

The primary considering thing is the battery usage. However, the discrete GPU will automatically enabled when I opening some applications, such as VirtualMachine or DownloadManager. Therefore, it will cause about 10Watts power consumption. As we all know that it is not necessary to enable discrete GPU when we are in the outside and without AC adapters.

In this reason, I have to monitor wether the discrete GPU are enabled.

So, could you please add some features that allowing user to see the current primary GPU intuitively?

exelban, 2020-12-26:
Hi.
Do you mean something like [this](https://github.com/exelban/stats/issues/232)?

Military-Doctor, 2020-12-26:
yes. I think I can close this issue.

closed 2020-12-26 by Military-Doctor