exelban/stats #254: Memory Pressure needle gone when pressure level is rising up to "4".

Military-Doctor, 2020-12-28:
**Describe the bug**
Memory pressure needle is missing when pressure level going to 4:

[image]

**Details:**
 - Device: Macbook Pro 16" 2019
 - MacOS: 10.15.7 (19H15)
 - Application version 2.4.9

Military-Doctor, 2020-12-28:
I found a [page](https://www.twblogs.net/a/5bc0d8032b717711c923f9bc) describes that: 

> In OS X 10.9 and later, you can also query kern.memorystatus_vm_pressure_level, which is a 1 (NORMAL), 2 (WARN) or 4 (CRITICAL)

exelban, 2020-12-28:
Hi. Thanks for reporting. I never saw another level then 1)

closed 2020-12-29 by Military-Doctor