exelban/stats #256: Feature Request: Monochrome colour scheme for upload/download arrows

JeffreyCA, 2020-12-29:
I find the flashing of the red and blue upload/download arrows a bit distracting. I would like a setting to always show the upload/download arrows in monochrome. What would be even better is the ability to set a colour scheme like the other menu items.

exelban, 2020-12-29:
Hi.

I'm not sure if it makes sense. If you don't want to see the "blinking" arrows. You can just hide it.

JeffreyCA, 2020-12-29:
True I can hide it, but I like to keep the arrows visible

exelban, 2020-12-29:
But these arrows (as dots and letters) are for indicating the network activity. If they will be monochrome, they will not indicate network activity.

JeffreyCA, 2020-12-29:
That's fair. For me at least, I prefer having the indicators as visual labels instead of to indicate network activity (I can always tell there's activity because the values are not 0).

closed 2021-01-03 by exelban