exelban/stats #257: Public IP is cut off in Network menu item if it's IPv6

JeffreyCA, 2020-12-29:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: MacBook Pro 16-inch
 - MacOS: 11.1
 - Application version: 2.4.10

It only shows 6 of the 8 "groups": **`aaaa:bbbb:cccc:dddd:eeee:fff`**`:gggg:hhhh`.

[image]

Some feature requests:
- Ability to choose between showing IPv4 or IPv6
- Or show *both* IPv4 and IPv6

exelban, 2020-12-29:
Hi. Thanks for reporting.
I will try to fix it.

exelban, 2020-12-29:
Maybe you have an idea of how it can be fit?)

JeffreyCA, 2020-12-29:
Would a smaller font work? Or can you make it always show IPv4? I don't know about others but I don't really care what my IPv6 address is.

exelban, 2020-12-29:
I think a smaller font is only one good solution for now(

There is no way to select between ipv4 or v6. An external IP address is fetching `https://api.myip.com`. It means that your internet provider gives them only v6 (or maybe v4 too, but the service decides to show only v6).

JeffreyCA, 2020-12-29:
In the future, maybe consider another API that can return both v4 and v6. E.g. https://www.ipify.org/ has endpoints for both.

exelban, 2020-12-29:
Does it return both v4 and v6 for you?
(PS: I was used this one, but my pi-hole blocked this service)

JeffreyCA, 2020-12-29:
Yes `https://api.ipify.org` returns my v4 and `https://api64.ipify.org` returns my v6 address.

exelban, 2021-01-06:
Hi.

I have changed the service for the IP to `https://api.ipify.org`. It's a temporary solution. Changing the font size will not fix the bug. Because the length of v6 is too big. And the font size must be very very small.

closed 2021-01-10 by exelban

JeffreyCA, 2021-01-10:
Thanks, confirming it's fixed

sam-anam1, 2021-02-02:
hello @exelban again 

I've been really waiting for reload button
it would be really great that has **reload icon** to recheck IP or could be submenu for **refresh**

its 'cause sometime IP change or connected to VPN that in need to copy the IP as clickable to copy to clipboard

my suggestion or maybe it could two icons 🔄   [image] beside as its clickable (to copy to clipboard)

exelban, 2021-02-02:
> hello @exelban again
>
> I've been really waiting for reload button
> it would be really great that has **reload icon** to recheck IP or could be submenu for **refresh**
>
> its 'cause sometime IP change or [...]

Please open a new issue if it's not related to this issue.