exelban/stats #260: GPU on Mac Mini M1

grandtom, 2020-12-29:
Hi,

I have the new Mac Mini M1 and I have a problem: GPU always displays 0%.

When I click on GPU, I can see 2 timelines: the second one seems to be the good one. I don't know why there is an unknown GPU.
[image]

Any idea ?

exelban, 2020-12-29:
Hi. It means that the temperature for this GPU is unavailable.

exelban, 2020-12-29:
Hi. Thanks for reporting. It's strange.
Could you try to switch `GPU to show` from `Automatic` to available GPU?

exelban, 2020-12-29:
The first one is temperature. And it's unavailable for this GPU.
`unknown` - because I don't know how to detect this GPU. I don't have m1 mac. Someone with m1 must help with this)

grandtom, 2020-12-29:
Thank you for your quick answer. I can't switch GPU to show (only Automatic). How can I help you ?

exelban, 2020-12-29:
Hmm, it's very strange. It looks like macOS says that this GPU is not active.

exelban, 2020-12-29:
Please run `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt` and post the output.

grandtom, 2020-12-29:
Here is the file.
[output.txt](https://github.com/exelban/stats/files/5751877/output.txt)

exelban, 2020-12-29:
Thanks. It helps very much.

grandtom, 2020-12-29:
You're welcome :)

MrBenFTW, 2020-12-31:
This is actually broken recently. I still have the source code from when this used to work. I will attach it along with a compiled build.
[Stats.zip](https://github.com/exelban/stats/files/5758001/Stats.zip)
[stats-master.zip](https://github.com/exelban/stats/files/5758002/stats-master.zip)

MrBenFTW, 2020-12-31:
[output.txt](https://github.com/exelban/stats/files/5758014/output.txt)

MrBenFTW, 2020-12-31:
@exelban I noticed you have moved it to waiting for next release? I just compiled the latest master branch 10 minutes ago and this is still an issue

exelban, 2020-12-31:
Thanks for the feedback.
Does it show GPU as Apple M1? Or still undefined?

MrBenFTW, 2020-12-31:
It shows as unknown. 
[image]
[image]
[image]

exelban, 2020-12-31:
Could you share the commit hash of your master?

MrBenFTW, 2021-01-01:
Sorry I was busy.

I downloaded it as a zip from master about 12 hours ago

exelban, 2021-01-01:
Thanks.

Could I ask you to run this build and post the log file?
This build will write some logs to the file `log.txt` which you can find in the `Documents`.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5759554/Stats.dmg.zip)

grandtom, 2021-01-02:
Hi @exelban 

Here is the log file.

exelban, 2021-01-02:
Thanks. Could you check this one?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5760155/Stats.dmg.zip)

grandtom, 2021-01-02:
@exelban 

Great job ! I have a "**i**GPU" (I don't know if it's OK)

[log 2.zip](https://github.com/exelban/stats/files/5760656/log.2.zip)

[image]

Feature request : Is it possible to hide the temperature if unavailable ?

exelban, 2021-01-02:
Thanks. 
`iGPU` means the integrated GPU. You can disable it with a toggle: `Show GPU type`.

I have the plan to redraw the whole GPU popup. I don't know when exactly I will do it. But in the near future.

grandtom, 2021-01-03:
Thanks @exelban for your work !  Have a nice day !

closed 2021-01-03 by exelban