exelban/stats #262: Missing 'top processes' in RAM popup

jromero2k, 2020-12-31:
**Describe the bug**

Updated to v2.4.10 from v2.4.9, now the 'Top processes' section of the "RAM" popup shows empty.

**Details:**
 - Device: MacBook Air mid-2012
 - MacOS 10.14.6
 - Application version 2.4.10

exelban, 2020-12-31:
Hi. Thanks for reporting.
I will try to fix it.

Could you maybe share the output of `top -l 1 -o mem -n 5 -stats pid,command,mem`?

jromero2k, 2020-12-31:
> Hi. Thanks for reporting.
> I will try to fix it.
> 
> Could you maybe share the output of `top -l 1 -o mem -n 5 -stats pid,command,mem`?

```
Processes: 323 total, 2 running, 1 stuck, 320 sleeping, 1893 threads 
2020/12/31 04:44:36
Load Avg: 2.51, 2.52, 2.50 
CPU usage: 6.87% user, 25.95% sys, 67.17% idle 
SharedLibs: 167M resident, 59M data, 35M linkedit.
MemRegions: 199264 total, 1749M resident, 53M private, 625M shared.
PhysMem: 8170M used (2372M wired), 21M unused.
VM: 1893G vsize, 1374M framework vsize, 307560045(0) swapins, 310616154(0) swapouts.
Networks: packets: 29439638/13G in, 25364450/4541M out.
Disks: 79995693/1779G read, 74065556/2958G written.

PID    COMMAND          MEM   
38091  Shrook           1946M+
25164  Brave Browser    649M+ 
0      kernel_task      517M+ 
69411  firefox          361M+ 
25187  Brave Browser He 355M+ 
```

closed 2021-01-03 by exelban