exelban/stats #266: Duplicate core temps, no GPU diode in sensors, incorrect CPU temp

20four80five, 2021-01-01:
Thank you @exelban for your Stats app.

Hopefully you can provide me with some assistance 🤞🏼 

**Bugs/requests**
1. **Sensors** > duplicate core temperatures
2. **Sensors** > missing GPU diode temperature / no option for GPU diode temperature
3. **CPU** > display CPU diode temperature

---

>1. **Sensors** > duplicate core temperatures

Within the **Sensors** section, it displays duplicate cores - I have an i5-8400 6-core CPU:
[image]

---

>2. **Sensors** > missing GPU diode temperature / no option for GPU diode temperature

The GPU temperature displays correctly but is it expected behavior to not have "GPU diode" within the Sensors section?

- correct GPU temperature:
[image]

- no option to display GPU diode temperature within the Sensors section:
[image]

---

>3. **CPU** > display CPU diode temperature

The **Sensors** is set to display CPU diode:
[image]

I have the "CPU diode" sensor selected and it shows the correct **CPU temperature**:
[image]

[image]

... however, within the **CPU section**, it does not display the "CPU diode" temperature, how can I change this so that the **CPU temperature** reads the "CPU diode"?
[image]

---

**Details:**

- Application version [v2.4.10]
- Device: Hackintosh / iMac (Retina 5K, 27-inch, 2019)
- macOS Big Sur 11.1
- CPU: Intel i5-8400
- Intel UHD 630 - headless mode
- GPU: AMD Sapphire RX 570
[image]

exelban, 2021-01-01:
Hi. Thanks for reporting.

1. Your SMC (FaceSMC or VirtualSMC) provides 2 types of temperatures for cores. It can be fixed only on your side.
2. You see the temperature of the GPU in the GPU module not from SMC. But from the GPU driver. So there is not `GPU diode`. It can be, see previous.
3. In the CPU popup not showing `CPU diode` sensor. But core 0 temperature (don't ask why). Which, I suppose, must be fixed.

20four80five, 2021-01-02:
Thanks for the reply.

>1. Your SMC (FaceSMC or VirtualSMC) provides 2 types of temperatures for cores. It can be fixed only on your side.

I used to use FakeSMC and have been using VirtualSMC for the past couple of months and have never had this duplication issue using iStat, HWMonitorSMC, or even the HWMonitor app from RehabMan's FakeSMC package.

For reference, here is the HWMonitor app using VirtualSMC:
[image]

...here is HWMonitorSMC app using VirtualSMC
[image]

> 2. You see the temperature of the GPU in the GPU module not from SMC. But from the GPU driver. So there is not GPU diode. It can be, see previous.

Are you suggesting that it can be added?

> 3. In the CPU popup not showing CPU diode sensor. But core 0 temperature (don't ask why). Which, I suppose, must be fixed.

Are you suggesting that this will be fixed on your end OR is this something that I need to fix?

Thanks

20four80five, 2021-01-02:
...Here is HWMonitor with FakeSMC
[image]

exelban, 2021-01-02:
3. Yes. I will fix it.
1. The app fetching all available sensors from SMC. And your SMC reports 2 temperatures `TC%c` and `TC%C`. Normally mac has only one of them.
2. I suppose you can take a look at ACPI.

closed 2021-01-10 by exelban