exelban/stats #271: [Feature request] Deleting update files

igorkulman, 2021-01-03:
**Describe the bug**
Would it be possible to delete update files after they are used? Currently in my downloads folder:

[image]

exelban, 2021-01-04:
Hi. Thanks.
I will take a look at this.

exelban, 2021-01-10:
This fix does not help(

I will move the new `.dmg` to the temporary folder. I don't want to do this in this release. Because it has already a few big changes. And this one could break something. And after it will be difficult to debug.

It must be fixed for the next release.

exelban, 2021-01-17:
Hi. It fixed already. But it will work only on the next update. Because the problem is in the update `initialization` process. And [this](https://github.com/exelban/stats/releases/tag/v2.4.13) release contains this fix. So the next update will not leave the `dmg`.

closed 2021-01-17 by exelban