exelban/stats #272: Feature request: Tearoff menus to windows

blackketter, 2021-01-03:
**Describe the bug**
Just discovered stats and I'm loving it.  Thanks! 

One thing that would be a great addition would be if the pulldown menus could be pulled down and kept open (or torn off) for those times when I want to keep an eye on the details for a longer period of time.  If you want to see an example, Fantastical does this well.

**Details:**
 - Device: MacBook Pro 2019
 - macOS Big Sur
 - Application version 2.4.11

exelban, 2021-01-04:
Hi.

Thanks for the proposal. I was thinking about this feature a few times.
I will make a little MVP with this feature. I'm just not sure if the app needs this.

closed 2021-01-10 by exelban

blackketter, 2021-01-10:
Just updated, this is fantastic.  Thanks @exelban!  A very solid MVP. 

I have a suggestion that the detached views have a close box.  Should I open a separate issue?

exelban, 2021-01-10:
Hi. I wanted to add a close button. But can't find a good design and place for it. If you don't have a proposition for a new design of popup. Where the close button could appear/dissapear. You could open a new issue with this. 
Otherwise, don't open a new issue.