exelban/stats #277: Add Airpods battery status

fernandomacho, 2021-01-06:
It would be possible to add the possibility of showing the battery of the devices we connect by bluetooth: keyboard, mouse, airpods, etc.

exelban, 2021-01-06:
Hi. Thanks for the proposal. 

I suppose it's possible. But not sure when I will find the time to implement this feature.
Also could be the problem of how to visualize this in the menu bar.

fernandomacho, 2021-01-06:
Hi, thoodfairy I think you have the visualization well resolved. I think istats menu also has this functionality with a Toodfairy-like display (https://c-command.com/toothfairy/)

vikassangwan1, 2021-01-08:
For Bluetooth Earbuds other than Airpods if you're interested. https://github.com/jariz/Akku
Only this app works. I haven't been able to find any other app on macOS which can display third party Bluetooth Earbuds' battery.

exelban, 2021-01-10:
Thanks for the examples.
I'm not sure where I could add this feature. I'm not sure if the Battery module is the right place for this.
Maybe some new module: `External` or something like this.

Or I could add this one into the Battery module. But in this case, I must redesign the popup view and the setting.

vikassangwan1, 2021-01-10:
A dedicated section to Bluetooth would be better. It displays the battery for your favourite 1 device in the menu bar. When you expand, it displays the battery of all connected devices such as iPhone, iPad etc.

closed 2021-07-11 by exelban

jack24254029, 2021-07-15:
I install v2.6.0, but I can't see the battery level of Magic Trackpad 2

P.S 藍芽 / 藍牙 mean Bluetooth on Traditional Chinese

[image]
[image]
[image]