exelban/stats #279: No removable drive is shown

mpl75, 2021-01-08:
**Describe the bug**
Even if Show removable disks is set On, no drive is shown in detail menu of Disk. I have Transcend JedDrive in my SD card slot. 

**Details:**
 - Device: MacBook Pro, retina, 13-inch, Early 2015 
 - MacOS: 11.1
 - Application version: 2.4.11

exelban, 2021-01-09:
Hi. Thanks for the reporting.

I suppose it's a similar problem to [this one](https://github.com/exelban/stats/issues/114).

Could you please share the output of the `system_profiler SPUSBDataType`?
Please run this command when the SD card is connected.

mpl75, 2021-01-10:
Hello,

here it is. If you need any more info, just tell me. Your soft is awesome! Do you want to create localization to Czech language?

Have a nice day
mp

2021-01-10 13:37:22.078 system_profiler[49171:4503571] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCIWPT
      PCI Device ID: 0x9cb1
      PCI Revision ID: 0x0003
      PCI Vendor ID: 0x8086

        Internal Memory Card Reader:

          Product ID: 0x8406
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 8.20
          Serial Number: 000000000820
          Speed: Up to 5 Gb/s
          Manufacturer: Apple
          Location ID: 0x14600000 / 10
          Current Available (mA): 900
          Current Required (mA): 896
          Extra Operating Current (mA): 0
          Built-In: Yes
          Media:
            SD Card Reader:
              Capacity: 129,79 GB (129 787 494 400 bytes)
              Removable Media: Yes
              BSD Name: disk2
              Logical Unit: 0
              Partition Map Type: MBR (Master Boot Record)
              S.M.A.R.T. status: Verified
              USB Interface: 0
              Volumes:
                SD card:
                  Capacity: 129,77 GB (129 770 717 184 bytes)
                  Free: 63,98 GB (63 984 635 904 bytes)
                  Writable: Yes
                  File System: Journaled HFS+
                  BSD Name: disk2s1
                  Mount Point: /Volumes/SD card
                  Content: Apple_HFS
                  Volume UUID: ED98D8FB-1B7C-3CFC-ADFF-E8726382689C

        Bluetooth USB Host Controller:

          Product ID: 0x8290
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 1.69
          Manufacturer: Broadcom Corp.
          Location ID: 0x14300000

        USB 2.0 Hub:

          Product ID: 0x0101
          Vendor ID: 0x1a40  (TERMINUS TECHNOLOGY INC.)
          Version: 1.11
          Speed: Up to 480 Mb/s
          Location ID: 0x14100000 / 4
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

            USB KB:

              Product ID: 0x624c
              Vendor ID: 0x05d5
              Version: 1.20
              Speed: Up to 1.5 Mb/s
              Manufacturer: KB
              Location ID: 0x14140000 / 15
              Current Available (mA): 500
              Current Required (mA): 500
              Extra Operating Current (mA): 0

exelban, 2021-01-10:
Thanks for the output. I will try to debug why it does not show an SD card. It looks strange because it must be(

I would like to add the Czech language to the app. You could take a look at how it could be done on some [PR](https://github.com/exelban/stats/pull/261).

mpl75, 2021-01-11:
Hi, I've created pull request (this is my first pull request ever :-) So I hope I did it correctly)
mp

closed 2021-01-17 by exelban