exelban/stats #282: Add close button to detached windows

blackketter, 2021-01-10:
As discussed in #272, I suggest that there be a way to close windows that have been torn off the menubar.  

I suggest that they should use the standard macOS window controls in the top-left of the new window.   The two buttons for Activity Monitor and the settings window could be relocated to the bottom of the window, like this:

[image]

This seems to be a common pattern for other menubar apps, like Fantastical.

Alternatively, the buttons could be put in the upper right corner, but this may appear too cluttered:

[image]

Also: Clicking on either the settings or Activity Monitor buttons should not close the window, leave that for the close button.

exelban, 2021-01-11:
Thanks for the propositions.
First option not acceptable at all. I will don't want to have different layouts for pinned and unpinned views.

The second one is a little bit better. But there will be only a close button. Also, it cannot be the default system buttons. It must be a custom one. Because: a) it will be not centered, b) it will close the whole window, and there is a different mechanism for close the popup

blackketter, 2021-01-11:
Gotcha.  I stole the buttons in the mockup from System Preferences which has all three buttons even though only two (close and minimize) will ever be active.  Minimize is probably not going to be used often, so a single close button is probably sufficient.  I was just trying to use the system default behavior.

The close button(s) could appear only when you drag the window away.  This would make the popup window appear less complicated in the typical use case.

Thanks!

closed 2021-03-07 by exelban

blackketter, 2021-03-07:
Works great, thanks!