exelban/stats #284: Add disk bandwidth history charts to popup

blackketter, 2021-01-10:
**Describe the bug**
The disk menu item can be set to display the current disk bandwidth used.  I'd like a history chart of this in the popup window (like network bandwidth).  Ideally there would be one charge for every disk displayed in the menu.

Maybe something like this:

[image]

**Details:**
- Device: MacBook Pro 2010
- MacOS 11.1.0
- Application version 2.4.12

exelban, 2021-01-11:
Thanks for the proposal. I'm not sure if it's necessary. 
For most people, it will be unnecessary information. Also, it will make the disk interface too complicated. Even I will be missing in the view.

blackketter, 2021-01-11:
Thanks, I understand.  

My example was extra complicated because I have several external disks attached.  I think most users will only have a single disk in that menu.

It's only one data point, but I'm also a user of iStats Menus and this is the only feature that I miss in Stats.  I often want to keep an eye on disk bandwidth over time to watch a disk intensive operation.

exelban, 2021-01-11:
I have 5(

exelban, 2021-01-17:
Hi. 
I check how it will look like. And for me, it's irrelevant information and takes too much space. I agree that it will be useful information for someone (I have not idea why, but still). But for most people, it will be unuseful and take look out from the spacing.

closed 2021-01-17 by exelban

blackketter, 2021-01-17:
Thanks.  I use it to keep an eye on large file operations and long executions to understand if there's a bottleneck or bad behavior.