exelban/stats #285: Top processes stop updating after a while

blackketter, 2021-01-10:
A minute or two after dragging CPU, Network, RAM or Battery menus to stay open, the top processes list stops updating.  The other stats and graphs do continue to update, only the Top processes are frozen.

**Details:**
- Device: MacBook Pro 2019
- MacOS 11.1.0
- Application version 2.4.12

exelban, 2021-01-11:
Hi. Thanks for reporting.
It happens when you change the focus from the popup to another window.

closed 2021-01-17 by exelban

blackketter, 2021-01-17:
I've just seen the Network popup window processes stop updating after a longer time.  It did continue to update after switching windows, but later on stopped.  The rest of the window appeared to be updating properly.

Version 2.4.13