exelban/stats #289: Battery/sensors module - different temperature values and wrong translation for polish language

yokonrad, 2021-01-13:
Hello. As i wrote in the subject - there are different battery temperatures in battery module and sensors module. I think sensors module is displaying correct battery temperature because its closer (or almost the same) to the temperature displayed by coconutBattery.

Also there is a wrong translation for polish language (or there isnt at all) in battery module in term "cycles". Its displaying english version "cycles" instead "cykli" or "cykle" (both of them are proper plural naming for term "cycles").

Do you need any help with translation to polish language?

exelban, 2021-01-13:
Hi. Thanks for the feedback.
The temperatue in the battery module is taken from the battery. And in the sensors from the SMC. And it could be a little bit different. But in theory it must be the same. I'm not supposing that macbook has 2 different battery temperture sensors. But it could be.

There is no translation `Cycles` for polish language. I will add it.

yokonrad, 2021-01-13:
Temperature in battery module on my MB is almost all the time between 29-31 degrees. It doesnt matter if macbook is cold or not. But the temperature in sensors module is different if the MB is cold or at home temperature. So i think temperature taken from the SMC is better.

exelban, 2021-01-17:
Hi. I add missing translations to the app.

About the battery temperature. I'm not sure what to do with this. I suppose that the sensors module could show one temperature (from SMC). And battery module the another (because it from macOS battery power management). In this case, you could check what SMC says, and what PM says.

closed 2021-01-17 by exelban

yokonrad, 2021-01-18:
Maybe include both of them in both sensors but set them names to `Temperature (SMC)` and `Temperature (PM)`? I dont have better idea. I think it shouldn't stay that way as it's a bit misleading.