exelban/stats #295: Crash after clicking on menu item while network pop up was open

blackketter, 2021-01-17:
Has only happened once, so far. The network window had been open for maybe about an hour. Crash log attached:

[Stats_2021-01-17-141703_lucky13.crash.txt](https://github.com/exelban/stats/files/5826928/Stats_2021-01-17-141703_lucky13.crash.txt)

Military-Doctor, 2021-01-18:
Hi, I have the same problem. I opened RAM panel for about 2 hours. Crash happened when I click menu item rapidly.

Military-Doctor, 2021-01-18:
Stats version: `2.4.12`
macOS version: `Big Sur 11.1 (20C69)`
Device: `MacBook Pro 16" 2019`

exelban, 2021-01-18:
Hi. I will take a look at this.

exelban, 2021-01-18:
Thanks. Tomorrow I will leave my Mac for some time to reproduce this)

Military-Doctor, 2021-01-19:
I found a new bug when I try to locate this crash problem:

[image]

Military-Doctor, 2021-01-19:
Process:               Stats [55620]
Path:                  /Users/USER/Library/Developer/Xcode/DerivedData/Stats-cfdejxtulkwktkelyqltxhvpjuht/Build/Products/Debug/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.4.13 (190)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [55620]
User ID:               501

Date/Time:             2021-01-20 02:36:29.552 +0800
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Bridge OS Version:     5.1 (18P3030)
Anonymous UUID:        0D805E9E-5B71-4124-876A-05EABFCAB5EA

Sleep/Wake UUID:       2BF636DC-0080-4291-BBA3-43340AD531D8

Time Awake Since Boot: 39000 seconds
Time Since Wake:       24000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [55620]

External Modification Warnings:
Thread creation by external task.

Application Specific Information:
Fatal error: Index out of range: file Swift/ContiguousArrayBuffer.swift, line 444
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	0x00007fff2c73c034 closure #1 in closure #1 in closure #1 in _assertionFailure(_:_:file:line:flags:) + 484
1   libswiftCore.dylib            	0x00007fff2c73bb87 closure #1 in closure #1 in _assertionFailure(_:_:file:line:flags:) + 87
2   libswiftCore.dylib            	0x00007fff2c73b863 closure #1 in _assertionFailure(_:_:file:line:flags:) + 99
3   libswiftCore.dylib            	0x00007fff2c73b4d1 _assertionFailure(_:_:file:line:flags:) + 513
4   libswiftCore.dylib            	0x00007fff2c71d429 _ArrayBuffer._checkInoutAndNativeTypeCheckedBounds(_:wasNativeTypeChecked:) + 249
5   libswiftCore.dylib            	0x00007fff2c72214e Array.subscript.getter + 78
6   eu.exelban.Stats.Battery      	0x00000001053d2e55 closure #1 in Popup.processCallback(_:) + 2645 (popup.swift:282)
7   eu.exelban.Stats.Battery      	0x00000001053bdd50 thunk for @escaping @callee_guaranteed () -> () + 48

closed 2021-01-24 by exelban