exelban/stats #296: upstream throughput graph twice the real upstream throughput

meiser79, 2021-01-18:
**Describe the bug**
I run an upstream iperf3 test. The stats graph is approx. twice the upstream throughput than the real upstream throughput. See attached screenrecord.
[capture.zip](https://github.com/exelban/stats/files/5830364/capture.zip)

**Details:**
 - Device: Macbook Pro 2020 (13 inch)
 - MacOS 10.15.7
 - Application version 2.4.14

meiser79, 2021-01-18:
Much easier to see with constant UDP upstream traffic: `iperf3 -c ping-ams1.online.net -p 5208 -4 -u -b 5M`

exelban, 2021-01-18:
Hi.

As I see from the video, it shows almost the same values. Could you please show at what moment the bug appears?

meiser79, 2021-01-18:
Here the video for the UDP upstream throughput graph where you can see that it's twice the real traffic.
[capture-udp.zip](https://github.com/exelban/stats/files/5830728/capture-udp.zip)

exelban, 2021-01-18:
Thanks. Strange. When I run the command you post previously, I see correct values. Do you use some 3d-party ethernet controller?

meiser79, 2021-01-18:
No, I use the intregrated wifi adapter. On my other MacBook Pro 2015, I can't reproduce the issue.

On my MacBook 2020, a VPN is open all the time in the background. Which interfaces are used for the graph?

exelban, 2021-01-18:
You can specify which interface must be used. By default, it tries to detect the default interface from macOS.

meiser79, 2021-01-18:
Even if I specify "WLAN (en0)", it shows the doubled traffic.

exelban, 2021-01-18:
Could you compare the Stats throughput with some other app (Activity Monitor)?

exelban, 2021-01-18:
I'm pretty sure that the problem somewhere between. I just visualize the values reported by the system. Additionally very strange that you have exactly doubled values.

meiser79, 2021-01-18:
Activity Monitor shows 1,52 MB/s (approx. 12 Mbps) which is even higher than the stats graph.

exelban, 2021-01-18:
Do you have another way to check the throughput except iperf3?
Could you also check what `nettop` shows?

meiser79, 2021-01-18:
Just recognized that the iperf3 traffic is shown correctly in the "top processes" list.
nettop also shows the TX traffic correctly.

exelban, 2021-01-18:
Top processes are from `nettop`. 
It looks like throughput from the `iperf3` process is correct. But throughput on the interface is doubled.

meiser79, 2021-01-18:
Using a USB Ethernet adapter (QNAP QNA-UC5G1T), the graph is correct. So this is maybe related to the wifi adapter?

exelban, 2021-01-18:
It could be related not only to the adapter itself. I suppose it some bug (or maybe not a bug) in the macOS. Super strange doubling the value. 
It definitely does not relate to the Stats app)

meiser79, 2021-01-18:
Just checked with `eul`, it shows the iperf3 upstream traffic sent via wifi correctly.

I don't think it's related to iperf3 as using the speedtest under https://www.speedtest.net shows the same doubled value.

JeffreyCA, 2021-01-19:
I'm experiencing the same issue. Under "Top Processes" it reports the correct upload speed. Cyberduck also reports that value. But the total that Stats reports is double that value.

[image]

JeffreyCA, 2021-01-19:
It's only doubling when set to "Interface based". It reports the correct total when using "Process based"

exelban, 2021-01-19:
@meiser79 could you please check if switching to the `Process based` reader will solve the problem?

meiser79, 2021-01-19:
Yes, switching to "process based" shows the right upstream value. But from my point of view, it does not solve the issue.

exelban, 2021-01-19:
The issue is inside the adapter drivers. It returns the wrong value.

closed 2021-01-19 by meiser79

JeffreyCA, 2021-01-19:
So where should we report this issue then?

exelban, 2021-01-19:
I suppose it must be debugged a little bit more.
I [take the interface](https://github.com/exelban/stats/blob/2786909e375542a9230816885ccff4404a53f6b4/Modules/Net/readers.swift#L100-L128) and read the stats from it. Maybe there is another way to check interface statistics. And compare it.

JeffreyCA, 2021-01-19:
I found [this issue](https://developer.apple.com/forums/thread/100494) but it's for iOS

exelban, 2021-01-19:
It looks like the same issue.
Do you use VPN?

exelban, 2021-01-19:
I'm using the same way to get the data.

JeffreyCA, 2021-01-19:
> It looks like the same issue.
> Do you use VPN?

No I'm not.