exelban/stats #299: Index out of range when clicking Battery menu item rapidly

Military-Doctor, 2021-01-19:
**Describe the bug**
Open `Instruments`, collect memory usage of stats, open one detached Network panel.

Then, rapidly click battery menu item multiple times.

The crash will happened when `Instruments` collecting memory data.

**Details:**
 - Device: MacBook Pro 16" 2019
 - MacOS 11.1 Big Sur
 - Application version 2.4.13 (190)

Military-Doctor, 2021-01-19:
[image]

```
Process:               Stats [55620]
Path:                  /Users/USER/Library/Developer/Xcode/DerivedData/Stats-cfdejxtulkwktkelyqltxhvpjuht/Build/Products/Debug/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.4.13 (190)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [55620]
User ID:               501

Date/Time:             2021-01-20 02:36:29.552 +0800
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Bridge OS Version:     5.1 (18P3030)
Anonymous UUID:        0D805E9E-5B71-4124-876A-05EABFCAB5EA

Sleep/Wake UUID:       2BF636DC-0080-4291-BBA3-43340AD531D8

Time Awake Since Boot: 39000 seconds
Time Since Wake:       24000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [55620]

External Modification Warnings:
Thread creation by external task.

Application Specific Information:
Fatal error: Index out of range: file Swift/ContiguousArrayBuffer.swift, line 444
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	0x00007fff2c73c034 closure #1 in closure #1 in closure #1 in _assertionFailure(_:_:file:line:flags:) + 484
1   libswiftCore.dylib            	0x00007fff2c73bb87 closure #1 in closure #1 in _assertionFailure(_:_:file:line:flags:) + 87
2   libswiftCore.dylib            	0x00007fff2c73b863 closure #1 in _assertionFailure(_:_:file:line:flags:) + 99
3   libswiftCore.dylib            	0x00007fff2c73b4d1 _assertionFailure(_:_:file:line:flags:) + 513
4   libswiftCore.dylib            	0x00007fff2c71d429 _ArrayBuffer._checkInoutAndNativeTypeCheckedBounds(_:wasNativeTypeChecked:) + 249
5   libswiftCore.dylib            	0x00007fff2c72214e Array.subscript.getter + 78
6   eu.exelban.Stats.Battery      	0x00000001053d2e55 closure #1 in Popup.processCallback(_:) + 2645 (popup.swift:282)
7   eu.exelban.Stats.Battery      	0x00000001053bdd50 thunk for @escaping @callee_guaranteed () -> () + 48
8   libdispatch.dylib             	0x00007fff2018d5dd _dispatch_call_block_and_release + 12
9   libdispatch.dylib             	0x00007fff2018e7c7 _dispatch_client_callout + 8
10  libdispatch.dylib             	0x00007fff2019ab86 _dispatch_main_queue_callback_4CF + 940
11  com.apple.CoreFoundation      	0x00007fff2046e970 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
12  com.apple.CoreFoundation      	0x00007fff20430852 __CFRunLoopRun + 2731
13  com.apple.CoreFoundation      	0x00007fff2042f6ce CFRunLoopRunSpecific + 563
14  com.apple.HIToolbox           	0x00007fff286b46d0 RunCurrentEventLoopInMode + 292
15  com.apple.HIToolbox           	0x00007fff286b44cc ReceiveNextEventCommon + 709
16  com.apple.HIToolbox           	0x00007fff286b41ef _BlockUntilNextEventMatchingListInModeWithFilter + 64
17  com.apple.AppKit              	0x00007fff22c4cde9 _DPSNextEvent + 883
```

Military-Doctor, 2021-01-19:
The `numberOfProcesses` is 8, however, the list returns 14 elements!

exelban, 2021-01-19:
Hmm, very strange. It's almost impossible. I don't know how it's possible. The variable is the same for the number of top processes and views.

exelban, 2021-01-19:
Could you maybe check why it returns 14 elements if you running the app in the Xcode?

Military-Doctor, 2021-01-20:
I will check for that crash. It seems impossible that `list` have more elements than `processes`

Military-Doctor, 2021-01-20:
Traceingback through the call-stack, I found the error occurs when retrieving data from `/usr/bin/top`.

Sorry for a mistake is that It actually returns 16 elements, which doubled the `numberOfProcesses`

Here is the reason: **There are two pieces of data in the output pipe**

[image]

Military-Doctor, 2021-01-20:
I think I figured out why this happened.

It is because multiple thread try to access a shared variable `self.task`.

Basically it is a thread synchronization problem.

Military-Doctor, 2021-01-20:
The problem is located in `Modules/Battery/readers.swift`, line 176.

```swift
public override func start() {
        // ...

        // LINE 176
        DispatchQueue.global().async {
            self.task = Process()
            let pipe = Pipe()
            
            self.task?.standardOutput = pipe
            self.task?.launchPath = "/usr/bin/top"
            self.task?.arguments = ["-o", "power", "-n", "\(self.numberOfProcesses)", "-stats", "pid,command,power"]
            var counter = 0
            pipe.fileHandleForReading.readabilityHandler = { ... }
}
```

Military-Doctor, 2021-01-20:
## Here is an example similar to this question:

```swift
//
//  main.swift
//  test
//
//  Created by Junyi Hou on 1/20/21.
//

import Foundation

let group = DispatchGroup()

public class MyProcess{
    var id     : Int = -1
    var stdout : Int = -1
    init(_ n: Int) {
        id = n
        usleep(1000) // simulate a time-consuming operation
    }
}

public class A {
    private var task : MyProcess? = nil
    
    public func test(_ n: Int){
        group.enter()
        DispatchQueue.global().async {
            self.task = MyProcess(n)
            
            let pipe = 2*n // simulate an unrepeated random file descriptor
            usleep(1000)
            
            self.task?.stdout = pipe
            
            print("Task",self.task!.id, "write to", self.task!.stdout)
            group.leave()
        }
        
    }
}

var a = A()
for i in 1...100 {
    a.test(i)
}

group.wait()
print("Done")

```

Military-Doctor, 2021-01-20:
Let's have a look to the output, you will notice that `Task 54 write to -1` 

The reason is because that the execution order of multithreads is managed by system, one possible order that can cause `Task 54 write to -1` seems like that:
[image]

Military-Doctor, 2021-01-20:
Now, back to this Index out of range problem.

The problem is located in Modules/Battery/readers.swift, line 176.
[image]

exelban, 2021-01-20:
Hi. Sorry for the long response.
I think it's the best issue I ever had)

I'm very thankful for such a nice description of the bug. But there is one big problem. I already remove this part of the code yesterday( Because of #295.

I have a few questions about this. How it's possible that 2 processes write data to the same pipe. If the pipe is creating for each process. How it's possible that they share the pipe?

exelban, 2021-01-20:
Additionally, the task is reinitialized every time when the queue is fired.

Military-Doctor, 2021-01-20:
It take me some time to figure it out, they write to the same pipe isn't because they sharing a pipe but they sharing `self.task`. 

**Here is the working flow:** 

1. ThreadA and ThreadB creates `pipe` in the local-scope. (now we have are 2 pipes)
2. ThreadB assign `pipe` to `self.task.standardOutput`. (step 5, in the above diagram)

> At this time, `self.task.standardOutput` is pointed to ThreadB's pipe. 
> Meanwhile, **ThreadB haven't launch its task.**

3. ThreadA assign `pipe` to `self.task.standardOutput`. (step 9, in the above diagram)

> **WARNING**! `self.task.standardOutput` is pointed to **ThreadA's pipe**.
> 
> At this point, ThreadA and ThreadB have not started the task, but the pipe in `self.task.standardOutput` has become the pipe that created in ThreadA.

4. `self.task.launch()`, BOOM!

Because of this reason, both ThreadA and ThreadB will write data to the same pipe.

This is just one possible scenario. :)

By the way, why #295 is one big problem?

I think this detached windows feature is nice)

exelban, 2021-01-20:
Thanks. I spend some time on the nice diagram which you post previously and understand how it was possible)

Nope, detached popup it's not a problem) I like this feature too.
I just say that you spend a lot of time debugging this bug (and explaining it). When I already remove this code(

I also fix the problem when this function calling a few times(
Code is not pushed, because I must clean up it.

Military-Doctor, 2021-01-20:
Thank you for your quick response and I appreciate the attention you are giving to this issue. I also learned a lot ;)

Let's make progress together!

exelban, 2021-01-20:
Hi again.

I sit and take a look at why this code was in the project. 
I forget that `top` works differently when trying to get processes by `power`. 
And I remind myself that the `read` function does not work at all in this reader. And it's a reason why there is a global process.

So I have a question, do you have a good solution for this problem?)

exelban, 2021-01-20:
I know how hotfix: just limit the number of processes before call callback.
But it does not look like fixing the problem.

Military-Doctor, 2021-01-21:
@exelban In my patch, I add a NSLock() to keep thread-safe. It can prevent unordered execution flow.

Please check this commit: [https://github.com/Military-Doctor/stats/commit/3ad3b3153ef134f61ceb6b691ac43c445dd498c2](https://github.com/Military-Doctor/stats/commit/3ad3b3153ef134f61ceb6b691ac43c445dd498c2)

exelban, 2021-01-21:
Thanks. I solve it in a little bit different way.

closed 2021-01-21 by Military-Doctor