exelban/stats #301: [RFE] Use

stmty9, 2021-01-19:


exelban, 2021-01-19:
Hi. Thanks for the feedback.
What do you mean by `more types`?

There is no `real` memory.

stmty9, 2021-01-19:
[image]
You can show different types, they are explained in my initial comment via the link.
It's not a super high priority, but Docker is a real doozy, when it comes to how much memory it has vs 'real' memory (according to activity monitor)

exelban, 2021-01-22:
Hi. There are 2 problems with this.
First of all. Apple does not provide this type of ram in the `vm_statistics`. So I have no way to fetch it.
Second. I see no sense to show this type.

closed 2021-01-24 by exelban