exelban/stats #303: Feature Request: Option to reset upload/download bandwidth total everyday

emlimap, 2021-01-20:
At present stats keeps on counting the total uploaded/downloaded bandwidth until either the application or machine is restarted.

If you go for days without restarting the application or machine by sleeping & resuming the laptop, then the total would keep on going up. It would be useful to set the application to reset the stats every day. This is especially useful when you are working off a metered connection and have to keep an eye on bandwidth usage for the day.

At the moment I am using an application called bandwidth+ https://apps.apple.com/us/app/bandwidth/id490461369?mt=12 which resets the counter and starts over every day. The only reason I still keep this app for.

exelban, 2021-01-21:
Hi. Thanks for the proposal.

Maybe you have an idea of how it could be implemented? I mean the UX part.

colebob9, 2021-03-24:
> Maybe you have an idea of how it could be implemented? I mean the UX part.

I would personally want this option in the general Network configuration menu, and have a dropdown list for how often you want it to be reset (every day, week, month, etc).

For manual resets, you could have it similar to what is already in the network widget dropdown for the Public IP refresh button, just have a "Reset Usage" button next to the Details header.

exelban, 2021-05-23:
Hi. I have added a button in the popup view which allows resetting total bandwidth.
I hope it will be more useful than an everyday reset.

closed 2021-05-23 by exelban