exelban/stats #307: Feature request: IPv6 address(es) should be shown

moriczr, 2021-01-22:
**Feature request:**
I've seen in #257 that there was a problem showing IPv6 (GUA) address when it was fetched from the external API instead of the Public IPv4, however, I think it would be great if one could check the available IPv6 addresses using the dropdown menu.

I would suggest the IPv6 address(es) (_or any IP address in general_) to be grouped by active interfaces. 
When IPv6 is available the addresses can be checked locally on the OS, there is no need to fetch them from external sources (using `ifconfig` for example).

Example from an other implementation:
[image]

Also, note that there could be _deprecated_ IPv6 addresses(es) available. Any IPv6 address that has valid `pltime` is a preferred address and should be shown.

Preferred IPv6 addresses of `en0`
[image]

**Details:**
 - Device: [Macbook Pro 2016]
 - MacOS [11.1 (20C69)]
 - Application version [2.4.13]

exelban, 2021-01-22:
Hi. Thanks for the feedback.
I need a new UI to show IPv4 and IPv6 (or a few IPv6) and the same time.

exelban, 2021-03-03:
Hi.
I add an option to fetch and show IPv6 too. I have used `https://api64.ipify.org` to obtain an IPv6.
`ifconfig` not very good for v6 (does not show any IP on one of my mac). And does not resolve v4. So it's a good solution, but not complete. I will very glad if someone will add an option to obtain the v6 via `ifconfig`. But I will not spend time on this.

moriczr, 2021-03-05:
Hi!
If there are valid IPv6 addresses available, `ifconfig` should show them, so I guess the mac you've mentioned hasn't got IPv6 address.
Fetching from external API should be fine, however, the case when there's no IPv6 address available should be handled somehow on the widget as well.
Anyway, I'd like to suggest the following APIs to fetch Public IPv4 and the used global Pv6 (_note that there could be multiple "public" IPv6 address available at the same time on the same interface, but the API returns the one which has been used for the fetching_)
http://v4.ipv6-test.com/api/myip.php
http://v6.ipv6-test.com/api/myip.php
http://v4v6.ipv6-test.com/api/myip.php
from https://ipv6-test.com/api/
I've been using these for a while, they have been quite reliable so far.

Besides the IPv6 address, I still think that grouping the IP addresses by active network interfaces would be a great feature. Should I raise a separate Feature request for that?

closed 2021-03-07 by exelban