exelban/stats #311: Sensors Widget have a rendering problem

Military-Doctor, 2021-01-23:
**Describe the bug**
It seems sensors widget are not drawing at the full resolution. It is blur.

(When I put this widget into touchbar, it happened again!)

[image]

**Details:**
 - Device: MacBook Pro 16" 2019
 - MacOS 11.1
 - Application version 2.4.13

exelban, 2021-01-23:
Hi. Do you have the same bug with release build?

Military-Doctor, 2021-01-23:
This problem occasionally occurred in the previous release version, and I did not take it seriously.

Just now I found a way to reproduce this problem 100%:

1. Use MacBook Pro built-in Display (3072x1920), while using a 2k external screen.
2. Start stats on the external screen with 2k resolution.
3. You will find the text on the MacBook Pro built-in display becomes blurred.

[image]

Military-Doctor, 2021-01-23:
The bug is located in file `./StatsKit/helpers.swift`, public class `CAText`

```swift
public class CAText: CATextLayer {

    public init(fontSize: CGFloat = 12, weight: NSFont.Weight = .regular) {
        super.init()
        
        self.font = NSFont.systemFont(ofSize: fontSize, weight: weight)
        self.fontSize = fontSize
        
        self.allowsFontSubpixelQuantization = true

        // BUG IS HERE ⬇️
        self.contentsScale = NSScreen.main?.backingScaleFactor ?? 1
        self.rasterizationScale = NSScreen.main?.backingScaleFactor ?? 1
        // BUG IS HERE ⬆️

        self.foregroundColor = NSColor.textColor.cgColor
    }
    // ...
}
```

When I start stats, if the active monitor is the external, then scaleFctor will be set to `1`, resulting in a blur effect on the MacBook Pro built-in display.

exelban, 2021-01-24:
Hi. The bug is related to the resolution. It tried to draw properly the text for an external monitor. As a result, rendered text for monitor resolution does not correct for internal resolution.

Military-Doctor, 2021-01-24:
Maybe you can enumerate the monitors then set the scaleFactor to the largest one?

exelban, 2021-01-24:
It must be not the largest one, but the main. 
Also, I didn't noticed that bug with MacBook Pro and external 2560x1080.

exelban, 2021-01-24:
Ideally, it must be unique to each monitor.

Military-Doctor, 2021-01-24:
Hi, thanks for replying.

I have the same idea as you. However, my monitor is BenQ PD2500Q, which does not natively support HiDPI display.

Just today, in order to improve the display quality of the external monitor, I used a third-party script to **turn on the HiDPI** display for it, and I found that this rendering problem was also solved.

Debugging in Xcode found that both of the screens' `scaleFactor` has become 2.0.
Prior to this, the values ​​of `scaleFactor` on the MacBook Pro’s internal display and external display were 2.0 and 1.0 respectively.

MuhsinFatih, 2021-02-08:
> This problem occasionally occurred in the previous release version, and I did not take it seriously.
>
> Just now I found a way to reproduce this problem 100%:
>
> 1. Use MacBook Pro built-in Display [...]

For me it's the exact opposite. I have a 1440p monitor and a 1200p monitor attached to an egpu. It renders crisp on the built in retina screen, but it's blurred on the other two monitors

exelban, 2021-02-09:
Hi. I check this bug. It does not relate to the `CAText` scale. And it happened when one display has a different scale factor than the second one. I don't know if there is a solution to this. At least, I don't know it.

exelban, 2021-04-11:
Hi.
I will close this issue. The work will be continued in #374. Because the solution for both these issues is the same: rewrite sensors widget.

closed 2021-04-11 by exelban