exelban/stats #312: Missing toolTip and Localizations

Tai-Zhou, 2021-01-23:
https://github.com/exelban/stats/blob/8337e5c6e7fec465c1faad11791c8d835e20fc91/Modules/Net/popup.swift#L192-L194
https://github.com/exelban/stats/blob/f997094dc4c23fd9fa7c08b82335ff698c4ea772/Stats/Supporting%20Files/en.lproj/Localizable.strings#L147

It seems the toolTip for the `ssidField` is missing.

And for GPU popup, toolTips are not localized. (CPU and RAM popups have no problem)

[image]

https://github.com/exelban/stats/blob/28155fb9dfdee0721065c79f85ed6674716dbce5/Modules/GPU/popup.swift#L197
https://github.com/exelban/stats/blob/28155fb9dfdee0721065c79f85ed6674716dbce5/Modules/GPU/popup.swift#L214

I think these two lines should be changed to `LocalizedString("GPU \(id)")`.

https://github.com/exelban/stats/blob/28155fb9dfdee0721065c79f85ed6674716dbce5/Modules/GPU/popup.swift#L125
https://github.com/exelban/stats/blob/28155fb9dfdee0721065c79f85ed6674716dbce5/Modules/GPU/popup.swift#L251

And these two lines.

exelban, 2021-01-23:
Hi. Yes, it should be translated.

exelban, 2021-01-23:
`ssidField ` does not have toopTip (`Click to copy wifi name`). Because you cannot copy wifi name by click)

Tai-Zhou, 2021-01-24:
> `ssidField ` does not have toopTip (`Click to copy wifi name`). Because you cannot copy wifi name by click)

But when I click the wifi name, it is indeed copied to my clipboard. 😵 (latest version: 2.4.13)

Military-Doctor, 2021-01-24:
I took a look at the code, and there was indeed no code do the copying to the clipboard.

But I can also copy the wifi name to the clipboard. 

I was thinking that not only the wifi name can be copied, but other fields can also be copied. 

Then I tried it and found that it really is. We can copy everything. 👏

exelban, 2021-01-24:
Yes, you are right. Everything can be copied to the clipboard. I will fix it)

exelban, 2021-01-24:
Ok, I tried to fix it with a quick fix. But it needs more changes. For now, it will be like this. Only IP's and mac will have a tooltip. I will remove the option to copy other values with time.

closed 2021-01-24 by exelban