exelban/stats #313: Feature Request: Smooth Chart

Military-Doctor, 2021-01-23:
Hi, it's me again )

If the CPU usage history chart can be smooth, then statistics should be better look!

Could we have this feature?

I started a new branch and did some work on it.

Left: Smoothed
Right: Original

[image]

[image]

Military-Doctor, 2021-01-23:
[image]

exelban, 2021-01-23:
Hi. Thanks. I'm not sure if it's better. Because a lot of information is missing when smoothing the chart.

Military-Doctor, 2021-01-23:
Maybe we could give users an option to enable smooth charts?

Military-Doctor, 2021-01-23:
Hi, thanks for replying.

What do you think of this? The user can choose whether to open the smooth chart in settings page. 
After all, some users may prefer visual smoothness. 😺 

[image]

I have tested this feature and am ready to submit to the master branch.

exelban, 2021-01-24:
Hi. Thanks for the proposal. I'm 99% sure that this is a redundant option. The chart must show the usage. And if the chart will be smoothed, the information about usage in short periods of time will be loosed. 

Also, it cannot be just a toggle. A future like this must be realized as a slider. Where you can specify the level of smoothness for each chart individually.

exelban, 2021-02-21:
Please reopen this issue, if you still want to add this feature to the app.

closed 2021-02-21 by exelban

Military-Doctor, 2021-03-01:
I am sorry for the late reply. Sincerely speaking that I think you're right, this is a redundant feature. I will reopen this issue if someone needed this feature later. 😃