exelban/stats #315: Crash MacBook Pro M1

raffaelepojer, 2021-01-24:
**Stats crash at startup**
I'm not entirely sure if Stats is to blame, but after startup the pointer freezes and I have to turn the computer off and on again. In the crash report it says that Stats crashed a few seconds after booting.

**Details:**
 - Device: MacBook Pro M1, 16 GB Ram
 - MacOS 11.1
 - Application version 2.4.13

Here the crash report:

```
Process:               Stats [675]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.4.13 (190)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [675]
User ID:               501

Date/Time:             2021-01-23 10:51:25.100 +0100
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Anonymous UUID:        52691D8F-7826-3F07-7704-1D4EEB40EF6F

Time Awake Since Boot: 36 seconds

System Integrity Protection: enabled

Crashed Thread:        9  Dispatch queue: com.repeat.queue

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000112d34000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [675]

VM Regions Near 0x112d34000:
    mapped file                 112d20000-112d30000    [   64K] r--/r-- SM=ALI  Object_id=93abb9b1
--> 
    MALLOC_LARGE                112d60000-112d6c000    [   48K] rw-/rwx SM=PRV  

Application Specific Information:
dyld: in dlopen()

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00000001a5e37ce8 mach_msg_trap + 8
1   libsystem_kernel.dylib        	0x00000001a5e38098 mach_msg + 76
2   com.apple.CoreFoundation      	0x00000001a5f6be98 __CFRunLoopServiceMachPort + 380
3   com.apple.CoreFoundation      	0x00000001a5f6a36c __CFRunLoopRun + 1216
4   com.apple.CoreFoundation      	0x00000001a5f69740 CFRunLoopRunSpecific + 600
5   com.apple.HIToolbox           	0x00000001ada86678 RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00000001ada864a8 ReceiveNextEventCommon + 688
7   com.apple.HIToolbox           	0x00000001ada861d8 _BlockUntilNextEventMatchingListInModeWithFilter + 76
8   com.apple.AppKit              	0x00000001a8741da4 _DPSNextEvent + 868
9   com.apple.AppKit              	0x00000001a8740724 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1312
10  com.apple.AppKit              	0x00000001a873260c -[NSApplication run] + 600
11  eu.exelban.Stats              	0x000000010452a80c 0x104528000 + 10252
12  libdyld.dylib                 	0x00000001a5e8cf34 start + 4

Thread 1:: Dispatch queue: com.apple.root.background-qos
0   libsystem_kernel.dylib        	0x00000001a5e37c28 _kernelrpc_mach_vm_deallocate_trap + 8
1   libsystem_kernel.dylib        	0x00000001a5e397f4 mach_vm_deallocate + 88
2   eu.exelban.Stats.CPU          	0x000000010478782c 0x104784000 + 14380
3   eu.exelban.Stats.ModuleKit    	0x00000001045d0ec0 0x1045b8000 + 102080
4   eu.exelban.Stats.ModuleKit    	0x00000001045e0b60 0x1045b8000 + 166752
5   eu.exelban.Stats.ModuleKit    	0x00000001045bf464 0x1045b8000 + 29796
6   libdispatch.dylib             	0x00000001a5cc28bc _dispatch_call_block_and_release + 32
[... 555 lines cut ...]
SQLite page cache                  128K        2 
STACK GUARD                       56.2M       11 
Stack                             13.3M       11 
VM_ALLOCATE                       1184K       34 
__AUTH                             691K      158 
__AUTH_CONST                      9715K      314 
__CTF                               759        1 
__DATA                            5799K      315 
__DATA_CONST                      10.3M      325 
__DATA_DIRTY                       954K      117 
__FONT_DATA                          4K        1 
__LINKEDIT                       480.4M       16 
__OBJC_CONST                      1517K      134 
__OBJC_RO                         64.5M        1 
__OBJC_RW                         2544K        1 
__TEXT                           213.5M      340 
__UNICODE                          588K        1 
libnetwork                         128K        8 
mapped file                      220.8M       20 
shared memory                       80K        5 
===========                     =======  ======= 
TOTAL                              2.6G     1961 
TOTAL, minus reserved VM space     1.3G     1961 

```

exelban, 2021-01-25:
Hi.
Thanks for the crash report. I will try to fix it.

exelban, 2021-01-26:
Does it happen only when booting the mac?

raffaelepojer, 2021-01-26:
> Does it happen only when booting the mac?

Yes, it has happened three / four times so far (it has been installed for about a month). I don't know if it has anything to do with it, but I think it happens when I try to click on it as soon as the Mac is turned on

exelban, 2021-01-26:
According to the crash report, it's crashed because of `CACornerMask`. 
```
protocol witness for SetAlgebra.update(with:) in conformance CACornerMask (in Stats) (<compiler-generated>:0)
```

exelban, 2021-01-26:
I will remove all unnecessary corner radius from the app (there are no so many) in the next release. And will ask you to report if the next release still crashing.

exelban, 2021-01-26:
But I cannot promise that it will fix the problem. It looks more like some internal bug.

raffaelepojer, 2021-01-26:
> I will remove all unnecessary corner radius from the app (there are no so many) in the next release. And will ask you to report if the next release still crashing.

yes if it happens again I will report it

raffaelepojer, 2021-01-26:
> According to the crash report, it's crashed because of `CACornerMask`.
> 
> ```
> protocol witness for SetAlgebra.update(with:) in conformance CACornerMask (in Stats) (<compiler-generated>:0)
> ```

I realised that it crashed when the theme on the menu bar remained black instead of turning white.

closed 2021-01-31 by exelban