exelban/stats #318: Feature request: show "Memory Pressure" in RAM

richardtop, 2021-01-25:
Since macOS uses a lot of memory to cache frequently opened files, the memory pressure indicator is more relevant than knowing how much of the RAM is currently occupied.

It would be great if I could change the "RAM" indicator to display the memory pressure instead of the amount of used memory, similarly to the Activity Monitor.
[image]

exelban, 2021-01-25:
Hi. The app already showing the memory pressure.
You can even colorize the menubar widget depending on memory pressure.

richardtop, 2021-01-25:
How can I Put the memory pressure into the menu bar?

exelban, 2021-01-25:
You cannot put the pressure level into the menu bar. It's visible in the popup. Or can be visualized by color in chart widget.

exelban, 2021-01-25:
[image]
[image]

richardtop, 2021-01-25:
That looks like a visualization of the used memory, not the pressure (as the pressure is way lower)

exelban, 2021-01-25:
Yes. The chart is presenting used memory. But the color of the chart presenting the RAM pressure level.

richardtop, 2021-01-25:
Would it be possible to add a "memory pressure" chart? Also I like the idea of having blue for used memory and green for the memory pressure.

It's often that the memory pressure is the more important indicator, as only if the pressure is too high, the user has to do something (e.g. close the apps).

exelban, 2021-01-25:
I don't know how memory pressure can be visualized.

I agree that memory pressure is more important than memory usage in macOS.

exelban, 2021-01-25:
The best option for me to visualize the memory pressure level is:
[image]

richardtop, 2021-01-25:
How does the memory pressure look when viewed via the terminal?

exelban, 2021-01-25:
1, 2, 4 - pressure levels

exelban, 2021-01-31:
Hi. Any proposes of how it can be visualized in the menu bar?

richardtop, 2021-02-05:
Hi, I think the best is to get the pressure value out of the `memory_pressure` command and feed the memory pressure percentage from the `System-wide memory free percentage:` property.

E.g.:

```
\richard@Richards-MacBook-Pro ~ % memory_pressure
The system has 17179869184 (4194304 pages with a page size of 4096).

Stats: 
Pages free: 21520 
Pages purgeable: 35807 
Pages purged: 1955197 

Swap I/O:
Swapins: 3271498 
Swapouts: 3621839 

Page Q counts:
Pages active: 1384067 
Pages inactive: 1355579 
Pages speculative: 31236 
Pages throttled: 0 
Pages wired down: 785131 

Compressor Stats:
Pages used by compressor: 616157 
Pages decompressed: 16701642 
Pages compressed: 26508007 

File I/O:
Pageins: 31529331 
Pageouts: 166002 

System-wide memory free percentage: 66%

```

The actual pressure value could be determined by the 100% - 66% = 44%, which corresponds to the value displayed in the Activity Monitor:

[image]

I think having a double-digit value is a better indication of the actual pressure. However, having an option to select between the digital (1/2/3/4) or percentage-based indication could be a good idea.

I guess, the digital value is used to determined the color of the pressure graph (1- green, 2 - yellow, 3+ - red, or in a similar way).

exelban, 2021-02-07:
As I say, I have only pressure levels: 1, 2, or 4.
`sysctl kern.memorystatus_vm_pressure_level`.

richardtop, 2021-02-15:
Ok, my suggestion then is to show the pressure digit value in the menu bar and have it as an option.
I.e. instead of showing "RAM 71%" it would show "M.P. 3"

exelban, 2021-03-05:
Hi. Sorry for the long response.
It cannot be `M.P. 3`. It does not mean anything. Only a percentage could be used for this. I will check if your calculation from the `memory_pressure` is correct. If yes, I will could it to visualize the pressure level.

richardtop, 2021-03-06:
Thanks, I think that would be the best way to visualize.

exelban, 2021-04-10:
Hi. I finally find some time to take a look at this. 
Unfortunately, this method does not work (throw `memory_pressure `).

`System-wide memory free percentage` - reports the memory utilization. It's the same value that shows now in the menu bar.

richardtop, 2021-04-10:
Hi, that doesn't seem to hold on my MacBook Pro 16 2019:
[image]

exelban, 2021-04-10:
It could be different values. Because some of the elements could not be included in the stat.
But it's for sure does not relate to the pressure.

richardtop, 2021-04-10:
Okay, clear, I guess we can close this.

exelban, 2021-04-11:
Hi. I would like to implement a reader to visualize memory pressure. But I need an algorithm of how it could be calculated.

closed 2021-04-11 by exelban

deepjia, 2025-05-23:
> Hi. I finally find some time to take a look at this. Unfortunately, this method does not work (`memory_pressure`).
> 
> `System-wide memory free percentage` - reports the memory utilization. It's the same value that shows now in the menu bar.

@exelban `System-wide memory free percentage` reports memory pressure.
`sysctl kern.memorystatus_vm_pressure_level` turns to `2` when `System-wide memory free percentage` is less than 50%

deepjia, 2025-05-23:
`memory_pressure` can also apply real or simulate memory pressure on the system.

`sudo memory_pressure -l warn` will continuously apply real memory pressure on the system, until `memorystatus_vm_pressure_level` turns to 2, and `System-wide memory free percentage` will drop to 49%

vladkens, 2026-11-06:
While adding memory pressure to [macmon](https://github.com/vladkens/macmon/pull/120) I disassembled Activity Monitor (macOS 27.2) to settle this. It calls `memorystatus_get_level()`, computes `100 - level`, then reads `kern.memorystatus_vm_pressure_level` (1, 2, 4). The same level is in `sysctl kern.memorystatus_level`, no root needed, so pressure is `100 - kern.memorystatus_level`, and `memory_pressure`'s "free percentage" is its inverse. It does relate to pressure.

```asm
100013148 add x0, sp, #0x44
10001314c bl 0x100047c58 ; symbol stub for: _memorystatus_get_level
100013150 cbnz w0, 0x100013218
100013158 ldr w8, [sp, #0x44]
10001315c mov w9, #0x64
100013160 sub w2, w9, w8
...
100013194 add x0, x0, #0x3ef ; literal pool for: "kern.memorystatus_vm_pressure_level"
1000131a8 bl 0x100047fb8 ; symbol stub for: _sysctlbyname
```

ArkfallBravo, 2026-11-06:
Building on the Activity Monitor finding above: the warning/critical level itself comes from a simple ratio in xnu, which Stats can compute from data it already reads.

The `VM_PRESSURE_*` checks in `osfmk/vm/vm_pageout.c` test `A < k·(A+C)`, where A is uncompressed available memory and C is compressed pages. That's the same as `C/(A+C) > 1−k`. From `host_statistics64` (no sysctl, same on Intel and Apple Silicon):

- A = `free_count + active_count + inactive_count` (`free_count` already includes speculative)
- C = `compressor_page_count`

On Macs over 3 GB the divisors in `vm_compressor.c` put warning above 50% and critical above ~66%. So the ratio shows how close the system is to the next level, not just which level it's at.

I've built this as a small RAM widget ("CMPR 42%", tinted by the kernel's level), plus a popup row and a `$pressure.compression` text token. It's running in my fork. @exelban, would you take a PR, either as a widget or as a colorize/fill source for the existing RAM widgets? Happy to shape it however you prefer.