exelban/stats #319: Feature Request: show CPU Idle percentage

richardtop, 2021-01-25:
I suggest an option to choose whether to show the actually used CPU percentage or the idle percentage.
Since it's what's can be used is often more relevant (i.e. how much more CPU time can you actually use)
[image]

The idea is similar to "power reserve" indicator on some cars:
[image]

exelban, 2021-01-25:
Again. This feature already implemented. Please check what the app has before opening a new issue.

[image]

richardtop, 2021-01-25:
Hi sorry for being unclear, but I wasn't able to find the way to put the idle percentage to the menu bar. Although, the pie chart solves the issue for me.

exelban, 2021-01-25:
There will be only used value. You cannot switch between used and idle.

richardtop, 2021-01-25:
Is it possible to somehow add that to be shown? e.g. (100%-used) ?

exelban, 2021-01-25:
Nope. Application showing only `used` values.

richardtop, 2021-01-25:
Okay, feel free to close this issue if you don't consider it fit for the product, as it's not highly important to me either. It's kind of "nice to have"

exelban, 2021-01-25:
Yes, I agree that it would be nice to have. But it's not a very useful option. I tried to keep fewer settings as it's possible.

richardtop, 2021-01-25:
Yeah, then let's close it. The memory pressure issue is far more important to me personally.

closed 2021-01-31 by exelban