exelban/stats #320: Question

ALT-KIK, 2021-01-26:
I want to make use cases and class diagrams for this soft. Is there SRS and related docs? Are there existing diagrams?

exelban, 2021-01-26:
Hi.
Nope. There is no documentation or diagrams(

exelban, 2021-01-31:
Hi. Please reopen the issue, if you would like to help with it.

ALT-KIK, 2021-01-31:
I want to help with it.

ALT-KIK, 2021-02-05:
How do you think, is this project mature enough to make SRS document?

exelban, 2021-02-07:
What does mean if the project mature enough?

ALT-KIK, 2021-02-11:
So can I start to make it? (Forget my last message please).

exelban, 2021-02-11:
Could you please say more about the `SRS document`?

ALT-KIK, 2021-02-11:
It defines:
-A purpose of a project
-An overall description
-Specific requirements

https://en.wikipedia.org/wiki/Software_requirements_specification

exelban, 2021-02-21:
Hi. I take a look at SPS, and I'm not sure if this app needs it. It looks like docs, which need to be updated. And I'm pretty sure that I will have no time for that.

closed 2021-02-28 by exelban