exelban/stats #323: Feature request: show network speed beside the chart

matteocontrini, 2021-01-27:
As mentioned in https://github.com/exelban/stats/issues/189#issuecomment-767781168, it would be great to have an option to show the download/upload network speeds when the chart view is selected.

Thanks!

exelban, 2021-01-29:
Hi. Thanks for the proposal.
I think it could be added.

kgifford, 2021-02-13:
Also multiple stats for multiple network interfaces, if possible

exelban, 2021-02-13:
@kgifford Hi. It's not really related to this issue. I propose you open a new issue with this feature.

exelban, 2021-02-18:
Hi.
It's could be achieved by enabling 2 widgets: Speed and Chart.

[image]

closed 2021-02-18 by exelban